zephyrproject-rtos/sdk-ng #9: Toolchain build fails for ARM

ioannisg, 2018-07-13:
I tried to use the provided script to build the ARM toolchain, on master branch.
Build failed, with the following log:

Building arm
~/Documents/workspace/sdk-ng/build/build_arm ~/Documents/workspace/sdk-ng/build
  CLEAN log
  CLEAN build dir
  CONF  oldconfig
*
* Restart config...
*
*
* Toolchain options
*
*
* General toolchain options
*
Build Static Toolchain (STATIC_TOOLCHAIN) [N/y/?] (NEW) 

Add crosstool-NG version to --version output (SHOW_CT_VERSION) [Y/n/?] y
Toolchain ID string (TOOLCHAIN_PKGVERSION) [] 
Toolchain bug URL (TOOLCHAIN_BUGURL) [] 
*
* Tuple completion and aliasing
*
Tuple's vendor string (TARGET_VENDOR) [zephyr] zephyr
Tuple's sed transform (TARGET_ALIAS_SED_EXPR) [] 
Tuple's alias (TARGET_ALIAS) [] 
*
* Toolchain type
*
Type
> 1. Cross (CROSS)
  2. Canadian (CANADIAN)
choice[1-2]: 1
*
* Build system
*
  Tuple        (READ HELP!) (BUILD) [] 
  Tools prefix (READ HELP!) (BUILD_PREFIX) [] 
  Tools suffix (READ HELP!) (BUILD_SUFFIX) [] 
*
* Misc options
*
Enable nls (TOOLCHAIN_ENABLE_NLS) [N/y/?] n
*
* C compiler
*
Compiler
> 1. gcc (CC_GCC)
choice[1]: 1
Show gcc versions from
> 1. GNU (GCC_USE_GNU)
choice[1]: 1
Source of gcc
> 1. Released tarball (GCC_SRC_RELEASE)
choice[1]: 1
Version of gcc
  1. 8.1.0 (GCC_V_8_1_0)
> 2. 7.3.0 (GCC_V_7_3_0)
  3. 6.4.0 (GCC_V_6_4_0)
  4. 5.5.0 (GCC_V_5_5_0)
  5. 4.9.4 (GCC_V_4_9_4)
  6. 4.8.5 (OBSOLETE) (GCC_V_4_8_5)
choice[1-6?]: 2
Flags to pass to --enable-cxx-flags (CC_GCC_ENABLE_CXX_FLAGS) [] 
Core gcc extra config (CC_GCC_CORE_EXTRA_CONFIG_ARRAY) [] 
gcc extra config (CC_GCC_EXTRA_CONFIG_ARRAY) [] 
List of multilib variants (CC_GCC_MULTILIB_LIST) [] 
Link libstdc++ statically into the gcc binary (CC_GCC_STATIC_LIBSTDCXX) [Y/n/?] (NEW) 
[... 194 lines cut ...]
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-2 core C gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@374]
[ERROR]  >>        called from: do_gcc_core_backend[scripts/build/cc/gcc.sh@671]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc/gcc.sh@257]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@679]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      https://crosstool-ng.github.io/docs/known-issues/
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]  
[ERROR]  (elapsed: 17:26.01)
/home/ioannisg/Documents/workspace/sdk-ng/bin/ct-ng:232: recipe for target 'build' failed
make: *** [build] Error 2
~/Documents/workspace/sdk-ng/build

ioannisg, 2018-07-13:
FYI @nashif

nashif, 2018-07-13:
what os / disto are you building on?

ioannisg, 2018-07-13:
Ubuntu 17.04

galak, 2018-12-10:
Can you try again, made a number of updates.

galak, 2019-01-24:
Closing this as we are building for ARM w/CI.  If still an issue please re-open.

closed 2019-01-24 by galak