zephyrproject-rtos/sdk-ng #32: Need to unify release and tag names

galak, 2019-01-23:
We seem to use different "version" formats for file name, tag, etc.  Need to clean this up so its the same everywhere.

galak, 2019-01-23:
For example the URL to get the beta4 release is:

https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.10.0-beta4/zephyr-sdk-0.10-beta4-setup.run

```
 dirname: v0.10.0-beta4
filename: 0.10-beta4
     tag: v0.10.0-beta4
```

nashif, 2019-01-23:
yeah, I am aware of this, will address.

nashif, 2019-01-24:
beta5 now building hopefully with consistent versioning:

https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.10.0-beta5

nashif, 2019-01-24:
btw, the tag does have a v prefix, similar to what we do in zephyr, the directory name on github is the tag name but the version string in the sdk name and elsewhere is now consistent.

closed 2019-02-12 by galak