zephyrproject-rtos/sdk-ng #46: Build Fails on arm64 host

ric96, 2019-02-21:
```
linaro@linaro-developer:~/zephyr/samples/basic/blinky/build$ cmake -DBOARD=96b_carbon ..
CMake Deprecation Warning at /home/linaro/zephyr/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:1 (include)

-- Selected BOARD 96b_carbon
Zephyr version: 1.10.99
Parsing Kconfig tree in /home/linaro/zephyr/Kconfig
Using /home/linaro/zephyr/samples/basic/blinky/build/zephyr/.config as base
warning: HAS_DTS_SPI_DEVICE (defined at /home/linaro/zephyr/dts/Kconfig:38) has unsatisfied direct dependencies (HAS_DTS_SPI), but is currently being selected by the following symbols:
BOARD_96B_CARBON (defined at /home/linaro/zephyr/boards/arm/96b_carbon/Kconfig.board:8), with value y, direct dependencies SOC_STM32F401XE && <choice> (value: y), and select condition SOC_STM32F401XE && <choice> (value: y)
CMake Error at /home/linaro/zephyr/cmake/extensions.cmake:983 (message):
  No such file or directory: CMAKE_READELF: 'CMAKE_READELF-NOTFOUND'
Call Stack (most recent call first):
  /home/linaro/zephyr/cmake/compiler/gcc.cmake:14 (assert_exists)
  /home/linaro/zephyr/cmake/toolchain.cmake:38 (include)
  /home/linaro/zephyr/cmake/app/boilerplate.cmake:240 (include)
  CMakeLists.txt:1 (include)

-- Configuring incomplete, errors occurred!
linaro@linaro-developer:~/zephyr/samples/basic/blinky/build$ 
```

Debian Strech with updated cmake 3.9
the only issue with environment setup was that gcc-multilib wasn't installed.

marc-hb, 2019-07-14:
> I don't think our build system has ever been tested (or supported) on anything other that x86 workstations.

Leaving that grand goal aside:

> CMake Error at /home/linaro/zephyr/cmake/extensions.cmake:983 (message):
>  No such file or directory: CMAKE_READELF: 'CMAKE_READELF-NOTFOUND'

CMake can be complicated and the code in zephyr/cmake/ even more, however this particular error message is not. What was your toolchain?

> Debian Strech with updated cmake 3.9

You need cmake 3.13 now. There's also been a number of toolchain changes, you may want to give this a new try.

cfriedt, 2019-11-23:
The instructions for adding using an alternative deb repository for cmake that are in the [Getting Started Guide](https://docs.zephyrproject.org/latest/getting_started/index.html) unfortunately do not work for arm64 because KitWare does not provide binaries for arm64.

So you must manually build cmake yourself for arm64 or use the Dockerfile or deb found [here](https://github.com/cfriedt/cmake-debs/releases/tag/v3.15.2). 

Just in case anyone else runs into errors building on arm64, aside from a more recent cmake, I also needed to
```
sudo apt install -y python3-dev libffi-dev libz-dev libxml2-dev libxslt-dev libjpeg-dev
```
I also needed a more recent cargo and rustc version than what was in the main repositories and then added ~/.cargo/bin/ to my PATH. 
```
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
```

It's worth pointing out, that I'm doing this inside of the Crositini environment on an arm64 Chromebook.

For a bare-metal arm cross-compiler, you can use

```
sudo apt install -y gcc-arm-none-eabi
```

That also requires setting the CROSS_COMPILE environment variable (see the [Getting Started Guide](https://docs.zephyrproject.org/latest/getting_started/index.html)).

marc-hb, 2019-11-25:
Debian stable version 10 released on July 2019 ships CMake version 3.13.4-1 https://packages.debian.org/buster/cmake The current Zephyr requirement is 3.13.1. 

Debian stable releases are generally two years in the making and "very" stable. Debian also has a good reputation for in-place upgrades, hopefully crostini doesn't interfere with that?

This other Zephyr page offers a number of alternatives to install CMake (thanks @dbkinder)
https://docs.zephyrproject.org/latest/getting_started/installation_linux.html?highlight=cmake#cmake [*]

Also worth noting: https://chromium.googlesource.com/chromiumos/docs/+/master/containers_and_vms.md#Can-I-run-a-different-Linux-distro
> Of course! The full LXD command line is available, and the included images remote has lots of other distros to choose from. However, we don't test with anything other than the default container that we ship, so things may be broken when running another distro.

[*] trying to guess the future permalink: https://docs.zephyrproject.org/2.1.0/getting_started/installation_linux.html?highlight=cmake#cmake

galak, 2019-11-25:
Some if these issues about cmake should be moved to the zephyr repo and not the SDK.  As the SDK currently only supports x86 as a host env.

cfriedt, 2019-11-25:
> Debian stable releases are generally two years in the making and "very" stable. Debian also has a good reputation for in-place upgrades, hopefully crostini doesn't interfere with that?

Crostini is currently on Debian 9 (at least for my device).
```
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

galak, 2021-04-15:
Closing this as we now publish aarch64 releases of the SDK.

closed 2021-04-15 by galak