zephyrproject-rtos/sdk-ng #55: 64 bit print format specifiers not defined

SteveOss, 2019-03-12:
Have upgraded SDK from 0.9.5 to 0.10.0. Looks like PRId64 and friends are no longer defined
(PRId32 etc are) ? Am building zephyr off head for the ARM frdm_k64f.

galak, 2019-03-12:
> Have upgraded SDK from 0.9.5 to 0.10.0. Looks like PRId64 and friends are no longer defined
> (PRId32 etc are) ? Am building zephyr off head for the ARM frdm_k64f.

Can you provide a test or how to reproduce.  As the PRI macros if building with min-libc come from zephyr itself and not the toolchain (in lib/libc/minimal/include/inttypes.h).

SteveOss, 2019-03-12:
uint64_t i64 = 0;
uint32_t i32 = 0;
printf ("Val32 %" PRIu32, i32);
printf ("Val64 %" PRIu64, i64);

 error: expected ')' before 'PRIu64'

Essentially PRIu64 is not #defined in arm-zephyr-eabi/arm-zephyr-eabi/include/inttypes.h as
__int64_t_defined is not defined but not exactly sure where this should be set ?

galak, 2019-03-12:
what does your config file look like?

SteveOss, 2019-03-12:
# General config
CONFIG_NEWLIB_LIBC=y
CONFIG_POSIX_API=y
CONFIG_PTHREAD_IPC=y
CONFIG_FLOAT=y
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y
CONFIG_HEAP_MEM_POOL_SIZE=4096
CONFIG_MAIN_THREAD_PRIORITY=10
CONFIG_STACK_USAGE=y
CONFIG_INIT_STACKS=y
CONFIG_MAIN_STACK_SIZE=4096

# Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4_AUTO=y
CONFIG_NET_TCP=y
CONFIG_NET_ARP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_PKT_RX_COUNT=100
CONFIG_NET_PKT_TX_COUNT=200
CONFIG_NET_BUF_RX_COUNT=100
CONFIG_NET_BUF_TX_COUNT=200

# Network address config
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.2.30"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.168.2.1"
CONFIG_NET_CONFIG_MY_IPV4_NETMASK="255.255.255.0"

# Net management
CONFIG_LOG=y

# Network debug config
CONFIG_NET_LOG=y
CONFIG_STDOUT_CONSOLE=y

SteveOss, 2019-03-12:
Is the same config that was working for the 0.9.5 SDK

SteveOss, 2019-03-12:
Yes. Just updated SDK this morning.

SteveOss, 2019-03-12:
Easy to fix in my code with:

#ifndef PRId64
#define PRId64 "lld"
#define PRIu64 "llu"
#endif

Everything else compiles and runs fine. But think should really be defined.

galak, 2019-03-12:
> Is the same config that was working for the 0.9.5 SDK

Agreed, its a bug and will look into fixing it.

galak, 2019-03-12:
Looks like the issue isn't the SDK but how Zephyr invokes things.  Going to move this over to Zephyr.

Opened issue:

https://github.com/zephyrproject-rtos/zephyr/issues/14310

closed 2019-03-12 by galak