zephyrproject-rtos/sdk-ng #61: This does not look like a tar archive

wangnuannuan, 2019-04-10:
On centos , it can't uncompress SDK
```
 [|] Installing x86 tools...tar: This does not look like a tar archive
tar: Skipping to next header
tar: Archive contains obsolescent base-64 headers
[/] Installing x86 tools... [-] Installing x86 tools... [\] Installing x86 tools...tar: Error exit delayed from previous errors
[*] Installing x86 tools... 
[|] Installing arm tools...tar: This does not look like a tar archive
tar: Skipping to next header
tar: Archive contains obsolescent base-64 headers
[-] Installing arm tools... [\] Installing arm tools... [|] Installing arm tools... [/] Installing arm tools... [-] Installing arm tools... [\] Installing arm tools... [|] Installing arm tools... [/] Installing arm tools... [-] Installing arm tools... [\] Installing arm tools... [|] Installing arm tools...tar: Error exit delayed from previous errors
[*] Installing arm tools... 
[|] Installing arc tools...tar: This does not look like a tar archive
tar: Skipping to next header
tar: Archive contains obsolescent base-64 headers

```
I modify the command parameter `-xf` to `-jxvf` in `setup.sh`

galak, 2019-04-12:
What version of tar do you have on centos?

wangnuannuan, 2019-04-15:
```
~ $ lsb_release -a
LSB Version:    :base-4.0-amd64:base-4.0-ia32:base-4.0-noarch:core-4.0-amd64:core-4.0-ia32:core-4.0-noarch:graphics-4.0-amd64:graphics-4.0-ia32:graphics-4.0-noarch:printing-4.0-amd64:printing-4.0-ia32:printing-4.0-noarch
Distributor ID: CentOS
Description:    CentOS release 6.6 (Final)
Release:        6.6
Codename:       Final
```

wangnuannuan, 2019-04-15:
0.10.0

wangnuannuan, 2019-04-16:
```
 tar --version
tar (GNU tar) 1.13.25
Copyright © 2001 Free Software Foundation, Inc.
This program comes with NO WARRANTY, to the extent permitted by law.
You may redistribute it under the terms of the GNU General Public License;
see the file named COPYING for details.
Written by John Gilmore and Jay Fenlason.
```

PR #67 (2019-05-01): Fix tar expansion with older version of tar
Need to explicit pass flags to tar as older versions don't auto handle
the compression.

Fixes: #61
commit 342a162d (2019-05-01): Fix tar expansion with older version of tar

Need to explicit pass flags to tar as older versions don't auto handle
the compression.

Fixes: #61
closed 2019-05-01 by galak