zephyrproject-rtos/sdk-ng #62: Zephyr SDK's wewlib doesn't support long long specifiers in printf

pfalcon, 2019-04-20:
In a Zephyr application with CONFIG_NEWLIB_LIBC, I'm doing `printf("%lld\n", foo);` and get `ld` as output. Ditto for "%llu".

According to https://stackoverflow.com/a/32949522/496009, newlib's long long printf support is disabled by default, and must be enabled with `--enable-newlib-io-long-long` option to configure.

We should consider whether we want to enable that. But not having support for 64-bit value printing is weird.

(Report is based on samples/net/sockets/sntp_client's usage of `LOG_DBG("time: %lld", epoch_time);`. That doesn't work (what's printed is `time: ERR`). Heck, even printf("%lld") doesn't work.)

pfalcon, 2019-04-24:
@galak, @nashif: Any comments?

galak, 2019-04-24:
Let's create a build of the SDK with long long enabled and see the size comparison

galak, 2019-04-24:
> Let's create a build of the SDK with long long enabled and see the size comparison

Since we set `--enable-newlib-nano-formatted-io` today, we need to disable that as the nano-formatted-io doesnt support long long.

pfalcon, 2019-04-24:
> Let's create a build of the SDK with long long enabled and see the size comparison

Do you mean comparison for generated code size? Sounds good, thanks.

pfalcon, 2019-08-21:
@galak: This is now going to be a blocker for GoogleIoT, due to https://github.com/GoogleCloudPlatform/iot-device-sdk-embedded-c/commit/d215cc5dc046323dbcc63400c05f9a4696941960#diff-6e458435d7562e1520557ee84e6791bc . Let's please do something about this. (If I should do something about this, please let me know. I didn't look into Zephyr SDK build process, so it may take a while for me to achieve something.)

pfalcon, 2019-08-21:
> This is now going to be a blocker for GoogleIoT

Ticketed as https://github.com/atigyi/zephyr/issues/5

galak, 2019-08-21:
is there an existing sample or test that uses long long?

galak, 2019-08-21:
https://github.com/zephyrproject-rtos/sdk-ng/pull/101

Need to see how this impacts code size.

pfalcon, 2019-08-21:
> is there an existing sample or test that uses long long?

Do you mean in Zephyr tree?

Grepping, following tests/samples use it: samples/net/gptp, tests/net/ipv6. It's unclear how easy would be use them to test the functionality though.

And overall, there's a bit of hacking around it already in the codebase, e.g. (docstring for printk()):

~~~
 * Length attributes h, hh, l, ll and z are supported. However, integral
 * values with %lld and %lli are only printed if they fit in a long
 * otherwise 'ERR' is printed. Full 64-bit values may be printed with %llx.
~~~

We definitely can/should add a dedicated test for libc + %lld/%llu.

pfalcon, 2019-08-21:
> #101

Thanks! I'm off tomorrow, by otherwise, let me bootstrap SDK building env and try it.

galak, 2019-08-21:
> Thanks! I'm off tomorrow, by otherwise, let me bootstrap SDK building env and try it.

If you wait for CI to pass, I'll post the link to where you can get a binary.

pfalcon, 2019-08-21:
> If you wait for CI to pass, I'll post the link to where you can get a binary.

Definitely will help, thanks!

galak, 2019-08-21:
> Definitely will help, thanks!

builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/101/zephyr-sdk-0.10.3-pr-101-setup.run

galak, 2019-08-22:
@pfalcon realized that the change in PR #101 doesn't do anything because we enable CT_LIBC_NEWLIB_NANO_FORMATTED_IO.  Need to re-spin it.

pfalcon, 2019-08-23:
> Need to re-spin it.

Ok at https://github.com/zephyrproject-rtos/sdk-ng/pull/101#event-2574886297, I see a timestamp `datetime="2019-08-22T00:51:51Z"`

While `curl --head https://builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/101/zephyr-sdk-0.10.3-pr-101-setup.run` gives me:

~~~
...
last-modified: Thu, 22 Aug 2019 02:29:23 GMT
...
~~~

So, I assume that URL contains the new build, downloading.

pfalcon, 2019-08-25:
For reference, code size comparisons posted at: https://github.com/zephyrproject-rtos/sdk-ng/pull/101#issuecomment-524606258

closed 2019-12-11 by stephanosio