zephyrproject-rtos/sdk-ng #65: arc emsk's openocd debugger doesn't work in Zephyr SDK 0.9.5

vonhust, 2019-05-01:
**Describe the bug**
After Zephyr SDK updated to 0.9.5,  the openocd debugger of arc em_starterkit doesn't work correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=em_starterkit_em7d ..
3. make
4. west debug
5. see error

**Expected behavior**
the debugger cann't run because of error caused by openocd.

**Impact**
can't debug the em_starterkit hardware

**Screenshots or console output**
The log is as follow:

Using runner: em-starterkit
Open On-Chip Debugger 0.10.0+dev-gea2753a7-dirty (2018-11-02-23:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: arc-em.cpu tap/device found: 0x200044b1 (mfg: 0x258 (ARC International), part: 0x0004, ver: 0x2)
**Register `identity' is not found.**
in procedure 'init' 
in procedure 'ocd_bouncer'

Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* arc-em.cpu         arcv2      little arc-em.cpu         halted
Info : Listening on port 6333 for tcl connections
Info : Listening on port 4444 for telnet connections
GNU gdb (GDB) 7.12.0.20161010-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pokysdk-linux --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/wren/zephyr/samples/philosophers/emsk/zephyr/zephyr.elf...done.
Info : accepting 'gdb' connection on tcp/3333
Remote debugging using :3333
warning: while parsing target description (at line 6): Boolean fields must be one bit in size
**warning: Could not load XML target description; ignoring
openocd: src/server/gdb_server.c:1188: gdb_get_registers_packet: Assertion `reg_packet_size > 0' failed.
Remote connection closed**

**Environment (please complete the following information):**
 - OS: Linux/Debian
 - Toolchain (Zephyr SDK)
 - latest commit

**Additional context**
Add any other context about the problem here.

nashif, 2019-05-02:
@abrodkin are you working on a patch based on @sstallion suggestion above?

abrodkin, 2019-05-02:
@nashif we're on holidays till Monday but once back at work next week we'll implement suggested fix. Is that ok?

nashif, 2019-05-02:
@abrodkin sounds good.

@galak lets wait for this fix to openocd before we release next SDK

abrodkin, 2019-05-07:
@nashif OpenOCD works for ARC again. Finally. But apparently there were more than 1 breakage :)
See  https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/commits/topic-zephyr

@sstallion Could you please review my changes and if all OK then we may merge them in Zephyr's branch (I may create a pull-request if needed):
1. Discussed above fix:
    https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/commit/de1a88b5dd910dd49df475c27d98e255d871231f
2. Another fix to accommodate your other improvement :)
    https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/commit/15288ca0096c5b249f9517caa9ce211b3c89ca01
3. And yet another one:
    https://github.com/foss-for-synopsys-dwc-arc-processors/openocd/commit/2a5f5125ac8fa0e1359b6be03b209f9f5d1ade82

nashif, 2019-05-07:
@abrodkin thank you, will take a look.

sstallion, 2019-05-08:
> @galak <https://github.com/galak> sure, but first I wanted @sstallion
> <https://github.com/sstallion> to comment on that change while it is
> sitting in my tree and if he's OK I'll submit it upstream.

Sounds good. I'm taking a look now - I'll let you know if anything stands
out.

PR #79 (2019-06-05): openocd: Update for gdb server fix
Update openocd to pull in an openocd GDB server fix from the ARC guys

Fixes #65
commit dbc7efa4 (2019-06-05): openocd: Update for gdb server fix

Update openocd to pull in an openocd GDB server fix from the ARC guys

Fixes #65
closed 2019-06-05 by galak