zephyrproject-rtos/sdk-ng #66: Update GCC to some version supporting -mindirect-branch=thunk

lpereira, 2019-05-01:
This option has been added to GCC 8 and backported to GCC 7, and uses retpolines for both indirect jumps and calls.  This is required to complement CONFIG_RETPOLINE, in order to mitigate against Spectre v2 for code written in C.

galak, 2019-05-01:
Is this really an issue since we are using gcc-8.3?

lpereira, 2019-05-02:
-mindirect-branch=thunk has to be enabled (conditionally, of course) to mitigate against this issue in affected machines. I don't know if that's the case.

galak, 2019-05-02:
> -mindirect-branch=thunk has to be enabled (conditionally, of course) to mitigate against this issue in affected machines. I don't know if that's the case.

My point or question, was is this feature something that needs to be explicitly enabled in the GCC build or something we just get for free now that the SDK is using a gcc-8.3 code base?

lpereira, 2019-05-10:
AFAICT, it's always available for the targets that may require it.  I don't have the SDK here to test, but it might be a good idea to see if the compiler whines or not.  If it doesn't whine, then this ticket can be closed (and another one opened on the kernel repo to track it).

pabigot, 2019-06-27:
Looks like it's enabled for Intel architectures; that and s390 are the only GNUC targets that seem to support it.
```
lilith[301]$ /usr/local/zephyr-sdk-0.10.1/i586-zephyr-elf/bin/i586-zephyr-elf-gcc -mindirect-branch=thunk -c /tmp/x.c
lilith[302]$ /usr/local/zephyr-sdk-0.10.1/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc -mindirect-branch=thunk -c /tmp/x.c
lilith[303]$ /usr/local/zephyr-sdk-0.10.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -mindirect-branch=thunk -c /tmp/x.c
arm-zephyr-eabi-gcc: error: unrecognized command line option '-mindirect-branch=thunk'; did you mean '-findirect-inlining'?
```

stephanosio, 2020-04-10:
Closing this issue since we support GCC 8.3.0 now and the problem is resolved as far as the SDK is concerned (it is up to the user/Zephyr to enable this option).

closed 2020-04-10 by stephanosio