zephyrproject-rtos/sdk-ng #86: Failed to build zephyr-sdk ( sdk-ng )

jiapei100, 2019-08-03:
Hi, all:

My host environments:
```console
➜  ~ uname -a
Linux lvision-GT72-6QE 5.0.0-21-generic #22-Ubuntu SMP Tue Jul 2 13:27:33 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
➜  ~ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco
➜  ~ gcc --version
gcc (Ubuntu 8.3.0-6ubuntu1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I can successfully install zephyr-sdk release **Zephyr SDK 0.10.1**, but failed to build **zephyr-sdk for riscv32** with the following **ERROR** messages:

```console
➜  sdk-ng git:(master) ✗ ./go.sh riscv32           
Building riscv32
....../sdk-ng/build/build_riscv32 ....../sdk-ng/build
  CLEAN log
  CLEAN build dir
  CONF  defconfig
#
# configuration written to .config
#
  GEN   savedefconfig
[INFO ]  Performing some trivial sanity checks
[ERROR]  Don't set LD_LIBRARY_PATH. It screws up the build.
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_Abort[scripts/functions@487]
[ERROR]  >>        called from: CT_TestAndAbort[scripts/functions@507]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@56]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      https://crosstool-ng.github.io/docs/known-issues/
[ERROR]  >>
[ERROR]  >> NOTE: Your configuration includes features marked EXPERIMENTAL.
[ERROR]  >> Before submitting a bug report, try to reproduce it without enabling
[ERROR]  >> any experimental features. Otherwise, you'll need to debug it
[ERROR]  >> and present an explanation why it is a bug in crosstool-NG - or
[ERROR]  >> preferably, a fix.
[ERROR]  >>
[ERROR]  >> NOTE: You configuration uses non-default patch sets. Please
[ERROR]  >> select 'bundled' as the set of patches applied and attempt
[ERROR]  >> to reproduce this issue. Issues reported with other patch
[ERROR]  >> set selections (none, local, bundled+local) are going to be
[ERROR]  >> closed without explanation.
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]  
[ERROR]  (elapsed: 26080105:23.16)
make: *** [....../sdk-ng/bin/ct-ng:261: build] Error 1
```

Any suggestions?

Thank you very much.
Pei

galak, 2019-08-05:
Any particular reason that you are trying to build the SDK yourself.  The easiest way is to utilize the docker container that we build it in.

galak, 2019-11-07:
There have been changed pushed to make things work on Ubuntu 18.04

galak, 2019-11-27:
Going to close this PR as there hasn't been any feedback if the recent changes have addressed the issue.  Feel free to re-open if this is still an issue.

closed 2019-11-27 by galak