zephyrproject-rtos/sdk-ng #100: SDK installer should propose to set path and variant in .zephyrrc, so user could start building instantly

nashif, 2019-08-20:
At the end of installation, the SDK script should ask if we should store the location and variant in .zephyrrc  to allow immediate use of the SDK.

galak, 2019-10-03:
Provide a y/n question to allow updating `.zephyrrc` (create if it doesn't exist)

galak, 2019-10-03:
Here's what I've got working:

```
[galak@spiff sdk-ng]$ ./zephyr-sdk-0.10.3-setup.run 
Verifying archive integrity...  100%   All good.
Uncompressing SDK for Zephyr  100%  
Enter target directory for SDK (default: /home/galak/zephyr-sdk/): 
Installing SDK to /home/galak/zephyr-sdk
Creating directory /home/galak/zephyr-sdk
Success
 [*] Installing x86 tools... 
 [*] Installing additional host tools... 
Success installing SDK.

You need to setup the following environment variables to use the toolchain:

     export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
     export ZEPHYR_SDK_INSTALL_DIR=/home/galak/zephyr-sdk

Update/Create /home/galak/.zephyrrc with environment variables setup for you (y/n)? 
y
SDK is ready to be used.
```

closed 2019-10-04 by galak