zephyrproject-rtos/sdk-ng #109: bundled x86 assembler chokes on expressions involving division

gnuless, 2019-09-17:
GNU as normally supports arithmetic expressions on absolute symbols involving the `/` operator (for division), and on stock versions it works as expected.  In the below text, the *stock* assembler versions of GNU as that I've tested that are not from the SDK: the system assembler on my Ubuntu dev system (which comes from binutils-2.30) and one source-built from binutils-2.32.  The *SDK* assembler is any version of `i586` and/or `x86_64` that I've tested in the Zephyr SDK: I've specifically tested 0.9.5, 0.10.0 and 0.10.3.

Input file `test.s`:
```X=(512/4)

movl $X, %eax
```

Run through the *stock* assembler with `as -o test.o test.s`, then `objdump -d test.o` and you get:
```Disassembly of section .text:

0000000000000000 <.text>:
   0:   b8 80 00 00 00          mov    $0x80,%eax
```

Hex `0x80` being 128, which is indeed 512/4.

If you try to assemble with the *SDK* assembler, you'll get:
```
test.s: Assembler messages:
test.s:1: Error: found '
', expected: ')'
```

More troublingly, if you drop the parentheses from the expression so `test.s` is:
```

X=512/4

movl $X, %eax
```

Then the *SDK* assembler doesn't choke, it just silently produces the wrong results:
```
Disassembly of section .text:

0000000000000000 <.text>:
   0:   b8 00 02 00 00          mov    $0x200,%eax
```

Note the immediate constant `0x200` is 512.  There was no division.

This problem appears limited to the `/` operator.  Other operators that I've tried (`*`, `>>` etc.) work properly.

galak, 2019-09-17:
I tried this test on fedora and get:

```
[galak@spiff zephyr]$ objdump -d test.o

test.o:     file format elf64-x86-64

Disassembly of section .text:

0000000000000000 <.text>:
   0:	b8 00 00 00 00       	mov    $0x0,%eax

```

gnuless, 2019-09-17:
@galak Sorry, there was a formatting error in the description, which left off the `X=` bit (which is crucial).

galak, 2019-09-17:
@gnuless the 2.32 you built from source, what config options did you give it?

gnuless, 2019-09-17:
> the 2.32 you built from source, what config options did you give it?

Oh, just very vanilla.  `./configure --prefix=/home/charles/binutils-test` or similar.  On an Ubuntu 18.04.3 x86_64 system, then `make install` and test.

dcpleung, 2019-10-29:
Seems like binutils treats slashes as start of a comment on i386, from: https://sourceware.org/binutils/docs-2.33.1/as/i386_002dChars.html#i386_002dChars
> If the --divide command-line option has not been specified then the ‘/’ character appearing anywhere on a line also introduces a line comment.

closed 2019-11-07 by galak