zephyrproject-rtos/sdk-ng #110: Create architecture-specific SDKs

dbkinder, 2019-09-25:
Today's SDK contains tools for **all** supported architectures in one uninstallable package that's a very large (1,091MB) zephyr-sdk-0.10.3-setup.run download that, I've observed, taking 20 minutes to download.  Developers typically have a specific board they're using and don't need tools for other architectures.  Having an architecture-specific SDK would reduce the setup time (and clutter) on their development systems.

It appears that the ``go.sh`` script used to generate the SDK can create architecture-specific SDKs already, but we're using the "all" option to build the single combined SDK.  Can we investigate publishing the combined and architecture-specific SDKs.

dbkinder, 2019-09-25:
@nashif @galak Can we investigate publishing the current combined but also architecture-specific SDKs.

dbkinder, 2019-09-25:
@dcpleung Your thoughts?

dcpleung, 2019-09-25:
I have been thinking about this for the Xtensa toolchains. One way is to produce a main SDK which supports all the QEMU targets, and others are add-ons (which install to the same directory).

The CI is already able to build all configured toolchains, but the issue is in packages. Right now everything gets packaged into one tar.gz. So the work would probably focus on changing the packaging script.

pfalcon, 2019-09-25:
+1

> taking 20 minutes to download

Depending on circumstances, can take hours to download.

galak, 2019-09-25:
What's the expectation w/regards to the SDK tools packages for such a thing?

pfalcon, 2019-09-25:
> What's the expectation w/regards to the SDK tools packages for such a thing?

I believe we already provide a separate archive for (some kind) of sdk tools? Can be like that: to use SDK, one would need to download arch-specific toolchain + common tools packages. If that's found too cumbersome, well, I guess we can go for one package which include arch toolchain + (duplicated) common tools. That still should be quite an improvement re: download speed.

dbkinder, 2019-09-25:
Note there are a couple more SDK developer experience improvement issues entered: #99 and #100

nashif, 2019-09-25:
few options there that will force us to move away from how we are doing things now:
1. Distribute tools with additional script that fetches the desired architecture and installs it alongside the tools, more architectures and specialized toolchains can be added on top

2. @mbolivar proposed a few times to make the toolchains part of the modules and distribute them as zephyr modules, so basically you get them via 'west' with no need to go and fetch them individually. This however does mean we still need to download the whole thing unless there is a way to just get one toolchain.

3. Make each toolchain an installable archive that can be added and stacked based on needs.

mbolivar, 2019-10-02:
> @mbolivar proposed a few times to make the toolchains part of the modules and distribute them as zephyr modules, so basically you get them via 'west' with no need to go and fetch them individually. [...]

We can support this in various ways -- I think the easiest would be to add project groups to west similarly to how they're supported by repo, and add a group namespace that's not fetched by default, but could be fetched manually.

galak, 2019-10-04:
I've made a 0.11.0-alpha release with arch / toolchain specific pkgs as well as the full SDK.

https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.11.0-alpha

closed 2019-11-07 by galak