zephyrproject-rtos/sdk-ng #123: Cannot debug x86_64 qemu application with GDB

abrodkin, 2019-10-16:
I'm trying to debug Zephyr application built for x86_64 QEMU with GDB but with no luck.

Zephyr application is built that way:
```
mkdir build
cd build
cmake -DBOARD=qemu_x86_64 ../samples/hello_world/
make
make run
```
Note `make run` step is necessary as otherwise produced Elf (`./zephyr/zephyr.elf`) won't be usable in QEMU.

Now start QEMU with GDB server (basically adding `-S -s` to normal QEMU invocation in `make run`):
```
$ZEPHYR_SDK_INSTALL_DIR/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-x86_64 -nographic -net none -pidfile qemu.pid -serial   mon:stdio -smp cpus=2 -kernel ./zephyr-qemu.elf -s -S
```

And finally connect with GDB to QEMU:
```
$ZEPHYR_SDK_INSTALL_DIR/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-x86_64 /x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gdb zephyr-qemu.elf
...
Reading symbols from zephyr-qemu.elf...(no debugging symbols found)...done.

(gdb) target remote :1234
Remote debugging using :1234
warning: Selected architecture i386 is not compatible with reported target architecture i386:x86-64
warning: Architecture rejected target-supplied description
Remote 'g' packet reply is too long (expected 308 bytes, got 608 bytes): 0000000000000000000000000000000000000000000000006306000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
(gdb)
```

At this point nothing could be done like single-step etc.

dcpleung, 2019-10-24:
To debug x86_64, `build/zephyr/zephyr.elf` should be used with GDB instead of `zephyr-qemu.elf`.

Also, due to the fact that QEMU is running as x86_64 64-bit mode but `zephyr.elf` is of x32 ABI, GDB expects x32 ABI registers but gets full set of 64-bit registers. You will need to do `set architecture i386:x86-64` telling GDB to expect 64-bit registers.

There is currently no good solution for this except setting the architecture every time you run GDB. Hopefully, in the near future, when Zephyr is 64-bit ready, we can use the full 64-bit mode instead of x32.

dcpleung, 2019-10-24:
Hm... there actually is no documentation for x86_64 at all... let me ping the original author of the x86_64 arch.

dcpleung, 2019-10-28:
The problem should have been fixed on master now, as we are actually building for 64-bit (-m64) for `qemu_x86_64`, and no longer using x32.

abrodkin, 2019-10-28:
@dcpleung could you please check if it works on your side.
I just tried on the master branch and only with `set architecture i386:x86-64` helps me to connect to the QEMU.

dcpleung, 2019-10-28:
It worked on my desktop.

Which file did you use with GDB? It should be `build/zephyr/zephyr.elf`, and not `zephyr-qemu.elf`.

abrodkin, 2019-11-05:
Indeed if I use `./zephyr/zephyr/elf` it works:
```
# $ZEPHYR_SDK_INSTALL_DIR/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gdb zephyr/zephyr.elf
...
Reading symbols from zephyr/zephyr.elf...done.
(gdb) target remote :1234
Remote debugging using :1234
0x000000000000fff0 in ?? ()
(gdb) b main
Breakpoint 1 at 0x100000: file .../samples/hello_world/src/main.c, line 12.
(gdb) c
Continuing.

Thread 1 hit Breakpoint 1, main () at .../samples/hello_world/src/main.c:12
12              printk("Hello World! %s\n", CONFIG_BOARD);
(gdb)
```

closed 2019-11-05 by abrodkin