zephyrproject-rtos/sdk-ng #126: Failed to run "hello_world" sample when building for arm and using trunc() function (from libm)

alexeRadu, 2019-10-23:
I'm building the sample app "hello_world" to which I've added one line that uses trunc() function:

```c
#include <zephyr.h>
#include <sys/printk.h>
#include <math.h>

void main(void)
{
        volatile int i = trunc(34);
	printk("Hello World! %s\n", CONFIG_BOARD);
}
```
I'm building the sample for the KW41z board and when running I get an exception for the trunc() function.

The trunc() function can be found in libm.a library that comes with the SDK.

When I build the project using the ARM-EMBED toolchain the sample runs.
I've used the "7-2018-q2-update" and "8-2019-q3-update" toolchain variants from ARM-EMBED.

galak, 2019-10-23:
Did you enable CONFIG_NEWLIB_LIBC, for the libm.a support.  By default we use the minlibc in zephyr has NO libm support.

galak, 2019-10-23:
If I try test w/`CONFIG_NEWLIB_LIBC` things seem to build correctly.

alexeRadu, 2019-10-23:
Yes, you are right. It worked when using ```CONFIG_NEWLIB_LIBC```.

closed 2019-10-23 by alexeRadu

alexandru-porosanu-nxp, 2019-10-25:
I agree that using CONFIG_NEWLIB_LIBC solves the problem at hand. But I find it awkward that there's no feedback at link time that you're linking with the wrong library; what @alexeRadu noticed is that if using -lm and no newlibc, the hello_world compiles (and links) properly but when running  on the target (M0+ core), he gets an invalid instruction exception. 
This looks like a principle of least astonishment issue: the user will have no idea why his software is failing and no hint on how to track it down. I think there should be some sort of link-time warning when using -lm and minlibc. @galak what do you think, does this make any sense?

galak, 2019-10-25:
With the zephyr toolchain I get:
```
/opt/zephyr-sdk-0.10.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: ../app/libapp.a(main.c.obj): in function `main':
/home/galak/git/zephyr/samples/hello_world/src/main.c:13: undefined reference to `trunc'
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:97: zephyr/zephyr_prebuilt.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:558: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

with the arm embedded toolchain I get (2018-q4):

```
/opt/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: ../app/libapp.a(main.c.obj): in function `main':
/home/galak/git/zephyr/samples/hello_world/src/main.c:13: undefined reference to `trunc'
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:97: zephyr/zephyr_prebuilt.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:558: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

galak, 2019-10-25:
With 2019-q3:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       10760 B       512 KB      2.05%
            SRAM:        3824 B       128 KB      2.92%
        IDT_LIST:          40 B         2 KB      1.95%
/opt/gcc-arm-none-eabi-8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: ../app/libapp.a(main.c.obj): in function `main':
/home/galak/git/zephyr/samples/hello_world/src/main.c:13: undefined reference to `trunc'
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:97: zephyr/zephyr_prebuilt.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:558: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

So I feel like we are seeing a link error.

alexandru-porosanu-nxp, 2019-10-30:
@galak, yes, you are correct: just compiling the hello_world.c without using any supplementary linker options, you'll receive the 'undefined reference' error. On the other hand, forcing "-lm" through target_ld_options() and declaring extern double trunc(double) will yield the correct result, without using CONFIG_NEWLIBC. But I'm not sure why me and @alexeRadu have seen the weird invalid instruction in this case. 

So in short, we'll be using newlibc if we need trunc() or any other functions that are not available in minlibc.

galak, 2019-10-30:
> @galak, yes, you are correct: just compiling the hello_world.c without using any supplementary linker options, you'll receive the 'undefined reference' error. On the other hand, forcing "-lm" [...]

when you say correct result what do you mean?

alexandru-porosanu-nxp, 2019-10-30:
I meant that calling trunc() yields the expected result and no exception is triggered.

galak, 2019-10-30:
ok, let me know if you run into any other issues or cases where we can improve the developer experience.

alexandru-porosanu-nxp, 2019-10-30:
Sure, will do, thanks!