zephyrproject-rtos/sdk-ng #133: Add aarch64-zephyr-elf target

stephanosio, 2019-10-30:
Add aarch64-zephyr-elf target support, in preparation for Cortex-A AArch64 port.

galak, 2019-10-30:
> Add aarch64-zephyr-elf target support, in preparation for Cortex-A AArch64 port.

@stephanosio will you look at doing this?

stephanosio, 2019-10-30:
cc @carlocaione @timork @vladimir-podbrezsky

stephanosio, 2019-10-30:
> @stephanosio will you look at doing this?

I will try if I have some time to spare after finishing #132- that is, unless @carlocaione or someone else who is currently working on AArch64 port wants to take this.

carlocaione, 2019-10-30:
I could look at this but not high priority at the moment.

galak, 2019-10-30:
Here's a starting point for it https://github.com/zephyrproject-rtos/sdk-ng/pull/134

closed 2019-11-07 by stephanosio

stephanosio, 2019-11-12:
I used `aarch64-zephyr-elf-gcc` from #134 (included in 0.11.0-alpha-5) to test the Cortex-R AArch64 port by @carlocaione today; it built and ran tests successfully without any problems.