zephyrproject-rtos/sdk-ng #139: Failed to build zephyr-sdk (Error in MD5 checksums)

tluio, 2019-11-02:
Hello, everyone!
I downloaded zephyr-sdk-0.10.3-setup.run, moved it to ubuntu18.04 in vmware, and then added mod x to it. Then I tried running it by ./zephyr-sdk-0.10.3-setup.run, but I got the following error:
`Verifying archive integrity...Error in MD5 checksums: b1126471e95432b4ae5c3f27e0a4ba39 is different from 6fd9a58385305600e82c6caa5b23f218`
How can I fix the error?
Thanks for any help!

stephanosio, 2019-11-02:
I cannot reproduce this issue. Something must have gone wrong while either downloading or moving it from the host to the Ubuntu 18.04 guest on your machine.

Please check the following.
```
$ md5sum zephyr-sdk-0.10.3-setup.run
14cff67e4a2bb5b299f87dde1123de1b  zephyr-sdk-0.10.3-setup.run
$ sha1sum zephyr-sdk-0.10.3-setup.run
7babe41bf300cccc139936547fece05ded23e577  zephyr-sdk-0.10.3-setup.run
```

tluio, 2019-11-02:
> I cannot reproduce this issue. Something must have gone wrong while either downloading or moving it from the host to the Ubuntu 18.04 guest on your machine.
>
> Please check the following.
>
> ```
> $ [...]

Thanks for your help!
I have tried several ways: 
(1) Move it from the host to the Ubuntu guest by mouse dragging.  It didn't work immediately.
(2) Move it from the host to the Ubuntu guest by copying and pasting. It didn't work immediately, either.
(3) Send it to the guest by xftp. It worked.
They all worked after a while.
It seems that I should wait for some time after vmware shows that the transfer has finished.

closed 2019-11-02 by stephanosio

ujur007, 2021-02-09:
I have recently downloaded the setup.run file for the Zypher OS but I am getting the "error in MD5 checksum"

What is the option to avoid one more download?

galak, 2021-02-09:
> I have recently downloaded the setup.run file for the Zypher OS but I am getting the "error in MD5 checksum"
> 
> What is the option to avoid one more download?

which specific file are you downloading, what OS/distro version are you trying to install it on, and what exact error message are you seeing?

ujur007, 2021-02-09:
Hi,
I am using the getting started guide here on this link.

https://docs.zephyrproject.org/latest/getting_started/index.html

The file is the following. I am using Ubuntu 20.04.

> wget https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.12.2/zephyr-sdk-0.12.2-x86_64-linux-setup.run

When I make it executable with chmod +x "_filename_" and ./filename I get the error

> Verifying archive integrity...  100%  Error in MD5 checksums: XXXXX is different from XXXXX

galak, 2021-02-09:
Can you try:

`shasum zephyr-sdk-0.12.2-x86_64-linux-setup.run`

ujur007, 2021-02-09:
Yes! the following is the output

> 628e4570c318a578d5b3cab3d799cb9420be9dab  zephyr-sdk-0.12.2-x86_64-linux-setup.run

galak, 2021-02-09:
Looks like the file didn't download correct.  The sha should be:

`9f4a0419a5f998dbcaf98f5dd1b3aafcdeb74637  zephyr-sdk-0.12.2-x86_64-linux-setup.run`