zephyrproject-rtos/sdk-ng #155: Use SDK 0.11 for shippable

carlocaione, 2019-12-08:
A lot of work currently being done on zephyr needs SDK 0.11. We need to use SDK 0.11 for `shippable` otherwise we get failures (see https://github.com/zephyrproject-rtos/zephyr/pull/20263)

stephanosio, 2019-12-08:
For now, we are still in 2.1 feature merge freeze, so this will need to wait a bit.

zephyrproject-rtos/zephyr#20625 is going to add the latest SDK 0.11 alpha release to the CI image, so it can be used for testing (special handling is required in the build script to activate this though).

carlocaione, 2019-12-08:
Thanks for the pointer to the zephyr issue (I wasn't aware that there was a story already open). Not trying to push this now, just wanted to track the issue.

galak, 2020-01-31:
Closing this as we now have SDK 0.11.x in CI.

closed 2020-01-31 by galak