zephyrproject-rtos/sdk-ng #165: error when running ARC GDB in SDK on Ubuntu 19.10

andrewboie, 2019-12-17:
`/opt/zephyr-sdk/arc-zephyr-elf/bin/arc-zephyr-elf-gdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory`

Workaround:

```
 $ sudo apt install libpython2.7
```

Workstation setup instructions may need to be updated, or SDK build parameters adjusted.
I haven't tested whether this problem is specific to ARC's GDB in the SDK, or other platforms.

nashif, 2019-12-17:
@galak  do we still depend on python 2.7 in the new SDK?

stephanosio, 2019-12-18:
Zephyr SDK 0.10.3
```
$ ldd /opt/sdk/zephyr-sdk-0.10.3/arc-zephyr-elf/bin/arc-zephyr-elf-gdb
        linux-vdso.so.1 (0x00007ffec45c7000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f3ae3760000)
        libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f3ae31e3000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3ae2fc4000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3ae2c26000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3ae2835000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f3ae3964000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f3ae2618000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f3ae2415000)
```

Zephyr SDK 0.11.0-alpha-8
```
$ ldd /opt/sdk/zephyr-sdk-0.11.0-alpha-8/arc-zephyr-elf/bin/arc-zephyr-elf-gdb
        linux-vdso.so.1 (0x00007ffddb6b1000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2c0d45a000)
        libpython3.6m.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0 (0x00007f2c0cdaf000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2c0cb90000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2c0c7f2000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2c0c412000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2c0c1fa000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2c0be09000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f2c0e00c000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f2c0bbd7000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f2c0b9ba000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f2c0b7b7000)
```

galak, 2020-01-08:
@andrewboie have you tried the 0.11.0-alpha-8?  Does it have this issue?

stephanosio, 2020-05-18:
Closing this since we have declared that Python 3.8 is required to run Python-capable GDB.

closed 2020-05-18 by stephanosio