zephyrproject-rtos/sdk-ng #168: qemu-sytem-x86_64 in Zephyr SDK 0.11.0 Alpha-8 fails to boot

stephanosio, 2020-01-02:
The `qemu-sytem-x86_64` included in the Zephyr SDK 0.11.0 Alpha-8 release fails to boot the Zephyr RTOS.

To reproduce this behaviour, point `ZEPHYR_SDK_INSTALL_DIR` to a Zephyr SDK 0.11.0 Alpha-8 installation and try running any tests/samples.

QEMU hangs after the `Booting from ROM..`:
```
[1/2] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu64,+x2apic
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20191210_190318-6871e12d79fb-zephyr
)
Booting from ROM..
```

Last known working version is Zephyr SDK 0.11.0 Alpha-7.

Note that the QEMU version was bumped from 4.2.0-rc1 in the Alpha-7 to 4.2.0-rc4 in the Alpha-8.

stephanosio, 2020-01-02:
TODO: Re-test the `qemu-sytem-x86_64` behaviour with the QEMU 4.2.0 release.

stephanosio, 2020-01-02:
Linux kernel boots on the Zephyr SDK 0.11.0 Alpha-8 `qemu-system-x86_64`, so the problem is likely with the Zephyr x86-64 port, not the QEMU itself.

```
/opt/sdk/zephyr-sdk-0.11.0-alpha-8/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-x86_64 \
-m 128 \
-cpu qemu64 \
-no-reboot \
-nographic \
-net none \
-serial mon:stdio \
-smp cpus=2 \
-kernel arch/x86/boot/bzImage \
-hda /dev/zero \
--append "root=/dev/zero console=ttyS0"
```

```
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20191210_190318-6871e12d79fb-zephyr
)
[    0.000000] Linux version 5.5.0-rc4+ (stephanos@DOMINUS) (gcc version 9.2.1 20191102 (Ubuntu 9.2.1-17ubuntu1~18.04.1)) #1 SMP Thu Jan 2 19:27:07 KST 2020
[    0.000000] Command line: root=/dev/zero console=ttyS0
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000007fdffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000007fe0000-0x0000000007ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.12.1-0-ga5cab58-dirty-20191210_190318-6871e12d79fb-zephyr 04/01/2014
...
```

The problem is likely that some (valid) changes, which are not compatible with the current Zephyr x86-64 port, were introduced in the QEMU 4.2.0 release.

stephanosio, 2020-01-02:
cc @andrewboie

andrewboie, 2020-01-02:
does disabling CONFIG_X86_MMU fix it?

andrewboie, 2020-01-02:
I can try to look into this next week. If there's any bisect point where it starts working that would be useful.

stephanosio, 2020-01-02:
> does disabling CONFIG_X86_MMU fix it?

No, still stuck at `Booting from ROM..` after setting `CONFIG_X86_MMU=n`.

stephanosio, 2020-01-03:
It turns out that there is nothing wrong with the QEMU or the SeaBIOS (bios.bin, bios-256k.bin). The culprit here is VGA BIOS (vgabios*.bin).

Offending QEMU commits:
https://github.com/qemu/qemu/commit/0221d73ce6a8e075adaa0a35a6ef853d2652b855
https://github.com/qemu/qemu/commit/0285747737417a7d154ddc3e5adcf1fd665566fd

The Alpha-8 QEMU boots successfully after `vgabios-stdvga.bin` is replaced with the one from Alpha-7.

Need to figure out what kind of sorcery is happening in the VGA BIOS that is causing direct kernel boot to fail.

pfalcon, 2020-01-03:
@stephanosio, mind that various flags passed to qemu affect what kind of BIOS binaries are used, and even fine-grained behavior. For example, we definitely had times when qemu didn't clear all screen on start (== seabios wasn't used?), and a case when BIOS or something else hosed PCI config/initialization in such a way that e1000 eth driver didn't work, which is again was fixed by playing with qemu machine type, etc. options.

stephanosio, 2020-01-03:
Bisected SeaBIOS:

Last good
https://git.seabios.org/cgit/seabios.git/commit/?h=rel-1.13.0&id=588eb12163b51c4d402383369456ce1cd1ec86ab

First bad
https://git.seabios.org/cgit/seabios.git/commit/?h=rel-1.13.0&id=083801db10b08d6d4e569d84f99a1398044eb483

stephanosio, 2020-01-06:
In summary, this bug is caused by the increased size of the VGA BIOS ROM image in the QEMU v4.2.0 release.

The following is a detailed analysis of this bug:

1. The size of the VGA BIOS ROM (in particular, `vgabios-stdvga.bin` that is used by default) was increased from 38KiB to 38.5KiB in the QEMU v4.2.0 release, which contains the Sea(VGA)BIOS v1.12 pre-release.
2. The direct multiboot kernel loader ROM (`multiboot.bin`) is loaded right after the VGA BIOS ROM and the increase in the VGA BIOS ROM size caused the load address of the multiboot kernel loader ROM to be changed from 0xC9800 to 0xCA000 (the SeaBIOS loads ROM images with 2KiB alignment).
3. The SeaBIOS boots to the entry vector function of the direct multiboot kernel loader BEV ROM and this function loads the kernel image to the memory using the QEMU `fw_cfg` interface.
4. The Zephyr x86-64 linker script specifies two separate memory regions (0x8000 and 0x100000) and the `fw_cfg` treats these as a large single region of 0x8000:(0x100000 + KERNEL_SIZE - 1), where all unused bytes are initialised with 0. Note that this area includes the BIOS and option ROM regions.
5. The multiboot kernel loader BEV ROM executes after the SeaBIOS completes system initialisation (i.e. all option ROMs are loaded and shadow memory is configured) and therefore the single memory region loading, which writes to the BIOS and option ROM regions, should not normally be a problem (the PAM registers are configured with the BIOS and option ROM regions marked as read-only).
6. The `qemu-system-x86_64`, unlike `qemu-system-i386`, enables the KVMVAPIC mechanism by default, which loads an additional KVMVAPIC ROM (`kvmvapic.bin`).
7. The KVMVAPIC ROM is loaded after the direct multiboot kernel loader ROM and, when the size of the VGA BIOS ROM is 38KiB, its load address is 0xCA000.
8. The KVMVAPIC ROM is assumed to be loaded at 0xCA000 by the QEMU regardless of whether the SeaBIOS actually loads it at 0xCA000, and this memory region, usually occupied by the PAM which configures it as read-only, is overridden to be writable by the QEMU due to the inherent design limitations of the KVMVAPIC.
9. Note that the new load address of the direct multiboot kernel loader ROM is 0xCA000 when the size of the VGA BIOS ROM is 38.5KiB and this overlaps with the KVMVAPIC ROM region assumed by the QEMU, which is overridden to be writable and no longer governed by the PAM.
10. The direct multiboot kernel loader ROM, as part of the kernel image loading process, writes 0 to its own code memory that is marked as writable and this causes the subsequent instructions to fail.

The exact failure observed is as follows (in the multiboot kernel loader ROM code):
1. The kernel image is read into the guest memory through `FW_CFG_KERNEL_DATA` with `rep insb %dx, %es:(%edi)` where ES = flat data memory mapping entry in GDT, EDI = 0x8000, ECX = 0x10A000.
https://gist.github.com/stephanosio/bbd335bed7e5ef106f7a69014403a82e#file-trace-alpha8-L24523-L24557
2. (qemu quirk) The above instruction stops executing after EDI = (the address of the first byte of the next instruction) due to the corrupted code memory.
https://gist.github.com/stephanosio/bbd335bed7e5ef106f7a69014403a82e#file-trace-alpha8-L24565
3. The kernel entry point address is normally loaded after the kernel image data is loaded, but the instruction that sets the `fw_cfg` selector index to `FW_CFG_KERNEL_ENTRY` is corrupted and executed as something else.
https://gist.github.com/stephanosio/bbd335bed7e5ef106f7a69014403a82e#file-trace-alpha8-L24569-L24570
4. The rest of the ROM code executes and eventually jumps to an invalid entry point address due to the aforementioned failure.

stephanosio, 2020-01-07:
The QEMU does dynamically retrieve the KVMVAPIC ROM address:
https://github.com/qemu/qemu/blob/f4d8cf148e43d942ef1202071e0cd66ce40322e0/hw/i386/kvmvapic.c#L669

This value (2KiB aligned) is re-aligned to a 4KiB page boundary and used as the starting address of the writable memory region in the `vapic_map_rom_writable` function, in order to avoid creating subpages:
https://github.com/qemu/qemu/blob/f4d8cf148e43d942ef1202071e0cd66ce40322e0/hw/i386/kvmvapic.c#L612

Related QEMU commit:
https://github.com/qemu/qemu/commit/9512e4a9edef2d9b59eb10a03184cd90a34f62d5

For instance, if the KVMVAPIC ROM is reported to be loaded at 0xCA800, the `vapic_map_rom_writable` function re-aligns this value to `0xCA000` and uses it as the starting address of the writable memory region.

While this "imprecise writable memory region allocation due to page boundary alignment" issue may be resolved by changing the SeaBIOS ROM load address alignment to 4KiB (hence always in sync with the QEMU memory region alignment), if the KVMVAPIC is ever used, this is still a problem because the KVMVAPIC ROM will be overwritten by the direct multiboot kernel loader (this ROM region must be writable for the state storage):
https://github.com/coreboot/seabios/blob/f21b5a4aeb020f2a5e2c6503f906a9349dd2f069/src/std/optionrom.h#L55

While disabling the KVMVAPIC in the QEMU source code is also an option, this introduces a deviation from the standard QEMU behaviour and is therefore less than ideal.

The most practical and reasonable solution to this problem would be to update the Zephyr x86-64 QEMU launch sequence such that:

1. `locore` section is removed from the launch ELF executable and saved into a separate binary file.
2. QEMU is launched with the launch ELF executable as kernel and the `locore` bin file loaded at 0x8000.

closed 2020-01-08 by galak