zephyrproject-rtos/sdk-ng #173: Doubled FLASH usage when using SDK 0.11.0-beta-1 (C++ app on Cortex-M)

r2r0, 2020-01-22:
I have a small C++ test application build for qemu_cortex_m3 board.
For SDK 0.10.3 FLASH usage is reported as about 28K where for SDK 0.11.0-beta-1 it is about 56K.

Is it expected beahviour?
Is it needed  to add some config options to get better results?

There is used the same project configuration for both builds.
Zephyr updated (west update) as today (ff1cd6e10c156930e271017551e5b0befb0c0f42).

It seems that the newer GCC does great job for code size optimization for application but the net result is not great.

[zephyr_10_3.map.txt](https://github.com/zephyrproject-rtos/sdk-ng/files/4096529/zephyr_10_3.map.txt)
[zephyr_11_0.map.txt](https://github.com/zephyrproject-rtos/sdk-ng/files/4096530/zephyr_11_0.map.txt)

When I compare map files, then everything looks very nice until there is reached a part with sections like: .text.malloc_extend_top, .text._malloc_r, .text._svfprintf_r, .text._dtoa_r, .text._realloc_r etc.
Some of these symbols are not present in the "old" SDK and some have much smaller size i.e.
malloc_r: 172 vs. 1732 bytes, svfprintf_r: 916 vs. 6224 bytes, realloc_r: 76 vs. 1620 etc.

jkiepert, 2020-01-24:
I can confirm this issue is also present on the ARC target. I see ~27kB increase in instruction memory usage and 1.4kB increase in data memory usage for a given C++ application. Memory usage is actually reduced with 0.11.0 when compiling a C-only application.

C-only App:
v0.10.3:
```
Memory region         Used Size  Region Size  %age Used
            ICCM:       19264 B       224 KB      8.40%
            DCCM:       20300 B       240 KB      8.26%
        IDT_LIST:         121 B         2 KB      5.91%
```
v0.11.0
```
Memory region         Used Size  Region Size  %age Used
            ICCM:       18148 B       224 KB      7.91%
            DCCM:       20300 B       240 KB      8.26%
        IDT_LIST:         121 B         2 KB      5.91%
```
C++ app:
v0.10.3:
```
Memory region         Used Size  Region Size  %age Used
            ICCM:      150796 B       224 KB     65.74%
            DCCM:      149412 B       240 KB     60.80%
        IDT_LIST:         905 B         2 KB     44.19%
```
v0.11.0:
```
Memory region         Used Size  Region Size  %age Used
            ICCM:      177980 B       224 KB     77.59%
            DCCM:      150808 B       240 KB     61.36%
        IDT_LIST:         905 B         2 KB     44.19%
```

galak, 2020-01-24:
Can you guys provide a test case / example / sample that shows this issue.

r2r0, 2020-01-24:
Here you are:
[cpp_test.zip](https://github.com/zephyrproject-rtos/sdk-ng/files/4108044/cpp_test.zip)

0.10.3:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       23744 B       512 KB      4.53%
            SRAM:        8236 B       128 KB      6.28%
        IDT_LIST:         120 B         2 KB      5.86%
```

0.11.0
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       51636 B       512 KB      9.85%
            SRAM:        9676 B       128 KB      7.38%
        IDT_LIST:         120 B         2 KB      5.86%
```

galak, 2020-01-24:
This is due to newlib changes in the new SDK.

The following tweak the Zephyr Kconfig build system should put things back to on par with SDK 0.10.3:

```
diff --git a/lib/libc/Kconfig b/lib/libc/Kconfig
index e4c60aac7a..574295f623 100644
--- a/lib/libc/Kconfig
+++ b/lib/libc/Kconfig
@@ -40,7 +40,6 @@ endchoice # LIBC_IMPLEMENTATION
 
 if NEWLIB_LIBC
 
-if "$(ZEPHYR_TOOLCHAIN_VARIANT)" = "gnuarmemb"
 config NEWLIB_LIBC_NANO
        bool "Build with newlib-nano C library"
        default y
@@ -48,7 +47,6 @@ config NEWLIB_LIBC_NANO
          Build with newlib-nano library, for small embedded apps.
          The newlib-nano library for ARM embedded processors is a part of the
          GNU Tools for ARM Embedded Processors.
-endif
 
 config NEWLIB_LIBC_ALIGNED_HEAP_SIZE
        int "Newlib aligned heap size"

```

Will make this change to Zephyr once SDK 0.11.0 is the default.

r2r0, 2020-01-24:
With this patch the cpp_test code gives now:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       21944 B       512 KB      4.19%
            SRAM:        7680 B       128 KB      5.86%
        IDT_LIST:         120 B         2 KB      5.86%

```

Also my sample C++ app from the first post shrunk FLASH by 2K and RAM by 0.5K
Comparing to 0.10.3:
FLASH 7% improvement
RAM 4.5% improvement

Impressive - thanks a lot.

galak, 2020-01-24:
Should hopefully be resolved by:

https://github.com/zephyrproject-rtos/zephyr/pull/22176

r2r0, 2020-01-30:
Resolved by 6317c82f063f1e1d165ac08c1cd2a0c3ea381f8d

closed 2020-01-30 by r2r0