zephyrproject-rtos/sdk-ng #175: Zephyr RTOS not detected by OpenOCD for v0.11.0 with ARC target

jkiepert, 2020-01-24:
OpenOCD reports 'Warn: No RTOS could be auto-detected!' and thus Zephyr thread information is not available:
```
Warn : No RTOS could be auto-detected!
Loading section text, size 0x4cf8 lma 0x0
Loading section sw_isr_table, size 0x408 lma 0x4cf8
Loading section devconfig, size 0xc0 lma 0x5100
Loading section rodata, size 0x42c lma 0x51c0
Loading section datas, size 0x380 lma 0x80000000
Loading section initlevel, size 0xc0 lma 0x80000380
Loading section _static_thread_area, size 0x240 lma 0x80000440
Loading section _k_mem_pool_area, size 0x1c lma 0x80000680
Loading section _k_mutex_area, size 0x14 lma 0x8000069c
Loading section _k_queue_area, size 0x10 lma 0x800006b0
Start address 0x26e8, load size 23724
Transfer rate: 227 KB/sec, 2156 bytes/write.
(gdb) 
```
prj.conf contains:
```
CONFIG_OPENOCD_SUPPORT=y
CONFIG_DEBUG=y 
```
I have confirmed Zephyr RTOS detection and thread aware debugging works fine when building OpenOCD from source from https://github.com/zephyrproject-rtos/openocd.git @ zephyr-20191003, but fails when building from https://github.com/zephyrproject-rtos/openocd.git @ zephyr-20200116 (which appears to be the OpenOCD version included in 0.11.0).

galak, 2020-01-24:
@EvgeniiDidin can you look into this?

EvgeniiDidin, 2020-01-28:
Hi @galak, @jkiepert,

For  zephyr-20191003 branch I added the patch (https://github.com/zephyrproject-rtos/openocd/commit/3d7db5535a56accd2d4b377b51fb8f3cfce59d0e), which moves **&hwthread_rtos,** to the bottom of the list. For some reason this patch is missing in zephyr-20200116. 

I guess we need to add this patch to **zephyr-20200116**  branch as well.

galak, 2020-01-28:
Any clue why this is needed?  No issue with the change, just not seeing anything obvious as to why.

galak, 2020-01-28:
Hopefully fixed by https://github.com/zephyrproject-rtos/sdk-ng/pull/177

EvgeniiDidin, 2020-01-29:
@galak , The time between the detection of the problem and its correction was little, so I even didn't think about creating the issue. Should I create one for tracking?

galak, 2020-01-30:
> @galak , The time between the detection of the problem and its correction was little, so I even didn't think about creating the issue. Should I create one for tracking?

Its not need right now, so don't worry about it.

jkiepert, 2020-02-03:
I can confirm that 0.11.1 solves this issue.

galak, 2020-02-03:
Fixed in SDK 0.11.1

closed 2020-02-03 by galak