zephyrproject-rtos/sdk-ng #179: Missing setjmp and longjmp in 32 bit libc for x86 (SDK 0.11.0)

r2r0, 2020-02-03:
It looks that symbols setjmp and longjmp are not defined in the 32bit version of libc.
As the result my project does not link properly (configured with CONFIG_NEWLIB_LIBC=y).

These symbols appear in the 64bit libc (as well as in all versions of ARM libraries).
Is it intentional?

```
hid@devel:/opt/zephyr-sdk$ nm -g x86_64-zephyr-elf/x86_64-zephyr-elf/lib/libc_nano.a | grep jmp
lib_a-setjmp.o:
0000000000000030 T longjmp
0000000000000000 T setjmp
hid@devel:/opt/zephyr-sdk$ nm -g x86_64-zephyr-elf/x86_64-zephyr-elf/lib/libc.a | grep jmp
lib_a-setjmp.o:
0000000000000030 T longjmp
0000000000000000 T setjmp
hid@devel:/opt/zephyr-sdk$ nm -g x86_64-zephyr-elf/x86_64-zephyr-elf/lib/32/libc.a | grep jmp
hid@devel:/opt/zephyr-sdk$ nm -g x86_64-zephyr-elf/x86_64-zephyr-elf/lib/32/libc_nano.a | grep jmp
hid@devel:/opt/zephyr-sdk$
```

galak, 2020-02-03:
This is a bug in the 32-bit library for x86.  I've pushed a PR that hopefully fixes the issue.

galak, 2020-02-04:
@r2r0 can you try this package out and see if it fixes your issue:

https://builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/180/zephyr-toolchain-x86_64-0.11.1-pr-180-setup.run

r2r0, 2020-02-04:
@galak it works partially.

It works if I use offcial 0.11.1 sysroots because with the one from the file you mentioned I cannot configure my project.

```
-- Zephyr version: 2.1.99
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "3.6") 
-- Selected BOARD qemu_x86
-- Found west: /home/hid/.local/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
CMake Error at /home/hid/prepilot/zephyr/cmake/host-tools.cmake:67 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /home/hid/prepilot/zephyr/cmake/app/boilerplate.cmake:443 (include)
  CMakeLists.txt:2 (include)

CMake Error at /home/hid/prepilot/zephyr/cmake/extensions.cmake:1457 (message):
  Assertion failed: The detected dtc version is unsupported.

      The version was found to be 0                                   
      But the minimum supported version is 1.4.6                        
      See https://docs.zephyrproject.org/latest/getting_started/                     
      for how to use the SDK's dtc alongside a custom toolchain.
Call Stack (most recent call first):
  /home/hid/prepilot/zephyr/cmake/host-tools.cmake:69 (assert)
  /home/hid/prepilot/zephyr/cmake/app/boilerplate.cmake:443 (include)
  CMakeLists.txt:2 (include)
```
I also cannot see nano version of the newlib in menuconfig.

galak, 2020-02-04:
Here's the full SDK link, can you try that.  Will have qemu as well.

https://builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/180/zephyr-sdk-0.11.1-pr-180-setup.run

galak, 2020-02-04:
> I also cannot see nano version of the newlib in menuconfig.

That's a different issue which needs addressing.

r2r0, 2020-02-04:
With mentioned full SDK pr-180 configuration and execution of my test works well.

newlib nano is still not visible in menuconfig (for qemu_x86 board). 
The only visible options related to newlib are about printf and scanf.

closed 2020-02-05 by galak

pfalcon, 2020-02-06:
I get missing setjmp/longjmp building JerryScript for qemu_x86 with SDK 0.11.1:

~~~
/mnt/hdd/opt/zephyr-sdk-0.11.1/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: ../../obj/lib/libjerry-core.a(js-parser.c.obj): in function `parser_raise_error':
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/parser/js/js-parser.c:2601: undefined reference to `longjmp'
/mnt/hdd/opt/zephyr-sdk-0.11.1/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: ../../obj/lib/libjerry-core.a(js-parser.c.obj): in function `parser_parse_source':
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/parser/js/js-parser.c:2060: undefined reference to `setjmp'
/mnt/hdd/opt/zephyr-sdk-0.11.1/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: ../../obj/lib/libjerry-core.a(js-scanner-util.c.obj): in function `scanner_raise_error':
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/parser/js/js-scanner-util.c:68: undefined reference to `longjmp'
/mnt/hdd/opt/zephyr-sdk-0.11.1/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: ../../obj/lib/libjerry-core.a(js-scanner-util.c.obj): in function `scanner_malloc':
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/parser/js/js-scanner-util.c:109: undefined reference to `longjmp'
/mnt/hdd/opt/zephyr-sdk-0.11.1/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: ../../obj/lib/libjerry-core.a(js-scanner.c.obj): in function `scanner_scan_all':
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/parser/js/js-scanner.c:2264: undefined reference to `setjmp'
/mnt/hdd/opt/zephyr-sdk-0.11.1/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/../../../../x86_64-zephyr-elf/bin/ld: /home/pfalcon/projects-3rdparty/Embedded/Zephyr/jerryscript/jerry-core/parser/js/js-scanner.c:3097: undefined reference to `setjmp'
~~~

So, perhaps this needs reopening.

galak, 2020-02-06:
@pfalcon this was fixed post 0.11.1, I haven't released a 0.11.2 with the fix yet.

pfalcon, 2020-02-07:
@galak:

> this was fixed post 0.11.1, I haven't released a 0.11.2 with the fix yet.

Makes sense, confirming that JerryScript builds w/o errors with https://builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/180/zephyr-sdk-0.11.1-pr-180-setup.run, so looking forward to the release.

r2r0, 2020-02-11:
@galak 
Do you have any plans for release date of 0.11.2 (with newlib longjmp fix)?

galak, 2020-02-11:
I'm hoping for end of this week.  Wanted to see if I can work on adding spec support for nano-newlib to work on x86 and other arch's that dont have support.

Is that something you could try and test out for me?

r2r0, 2020-02-11:
x86 on QEMU - no problem

PR #180 (2020-02-05): newlib: Fix 32-bit build of newlib for setjmp
The 32-bit build of newlib didn't include setjmp/longjmp due to us not
configuring it correctly.  In configure.host we needed to set
mach_add_setjmp=true for 'm32'.  Also mimic i.86 in setting
libm_machine_dir=i386.

Fixes #179
commit df84499e (2020-02-05): newlib: Fix 32-bit build of newlib for setjmp

The 32-bit build of newlib didn't include setjmp/longjmp due to us not
configuring it correctly.  In configure.host we needed to set
mach_add_setjmp=true for 'm32'.  Also mimic i.86 in setting
libm_machine_dir=i386.

Fixes #179