zephyrproject-rtos/sdk-ng #181: aarch64 Toolchain wont run on aarch64 Manjaro

evantaylor, 2020-02-04:
I may be completely wrong here, but I'm getting the following error, and I suspect it is improperly built arm64 binaries.

```
evantaylor at Pinebook-Pro in ~/Projects/zephyr/samples/hello_world/build (master●) 
$ cmake -DBOARD=hifive1 ..
-- Zephyr version: 2.1.99
-- Selected BOARD hifive1
-- Found west: /usr/bin/west (found suitable version "0.6.3", minimum required is "0.6.0")
CMake Error at /home/evantaylor/Projects/zephyr/cmake/compiler/gcc/generic.cmake:24 (message):
  Executing the below command failed.  Are permissions set correctly?

  '/opt/zephyr-sdk/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc --version'

Call Stack (most recent call first):
  /home/evantaylor/Projects/zephyr/cmake/generic_toolchain.cmake:70 (include)
  /home/evantaylor/Projects/zephyr/cmake/app/boilerplate.cmake:459 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
```
Running SDK 0.11.1 and Zephyr Master Branch.

with a modified generic.cmake to point to aarch64 binaries:

```
evantaylor at Pinebook-Pro in ~/Projects/zephyr/cmake/toolchain/zephyr/0.11 (master●) 
$ cat generic.cmake 
# SPDX-License-Identifier: Apache-2.0

set(TOOLCHAIN_HOME ${ZEPHYR_SDK_INSTALL_DIR})

set(COMPILER gcc)
set(LINKER ld)
set(BINTOOLS gnu)

set(CROSS_COMPILE_TARGET aarch64-${TOOLCHAIN_VENDOR}-elf)
set(SYSROOT_TARGET       aarch64-${TOOLCHAIN_VENDOR}-elf)

set(CROSS_COMPILE ${TOOLCHAIN_HOME}/${CROSS_COMPILE_TARGET}/bin/${CROSS_COMPILE_TARGET}-)
set(SYSROOT_DIR ${ZEPHYR_SDK_INSTALL_DIR}/sysroots/${SYSROOT_TARGET}/usr)
set(TOOLCHAIN_HAS_NEWLIB ON CACHE BOOL "True if toolchain supports newlib")
```

galak, 2020-02-04:
The pre-built toolchain is an x86-64 hosted toolchain so there is not any expectation that it would work on an aarch64 host.

evantaylor, 2020-02-04:
> The pre-built toolchain is an x86-64 hosted toolchain

OK, that makes more sense. 

> so there is not any expectation that it would work on an aarch64 host.

The documentation, README.MD, or install scripts should at least say that somewhere, or throw a warning if not running x86-64.

Will try to build the toolchain now, but have been running into problems with the build scripts.  I'll see what I can do over a weekend.

galak, 2020-02-04:
> > The pre-built toolchain is an x86-64 hosted toolchain
> 
> OK, that makes more sense.
> 
> > so there is not any expectation that it would work on an aarch64 host.
> 
> The documentation, README.MD, or install scripts should at least say that somewhere, or throw a warning if not running x86-64.

Can add something to the README.MD on this, will also look to rename the files to hopefully make it more obvious in the future.

> 
> Will try to build the toolchain now, but have been running into problems with the build scripts. I'll see what I can do over a weekend.

In theory this should work, but that's theory.  I'm happy to take a fixes needed to make it build.  I'd actually like to start publishing aarch64, and would be willing to start publishing builds / binaries for aarch64.  I'm happy to take help on this front.  We need to try and get things building and than see if we can get the docker container working for AARCH64 so we can do consistent builds/releases/etc.

evantaylor, 2020-02-05:
> Can add something to the README.MD on this, will also look to rename the files to hopefully make it more obvious in the future.
>
>
> In theory this should work, but that's theory. I'm happy to take a [...]

I'll dive in this weekend/next weekend and see where I get stuck.  We're evaluating Zephyr as our IoT platform for some environmental research applications (remote sensing platforms), so I'll first just get my AMD64 environment working correctly before I attempt the aarch64 platform.  Would really love to develop some I2C

> I'm happy to take help on this front. We need to try and get things building and than see if we can get the docker container working for AARCH64 so we can do consistent builds/releases/etc.

Pine64 just announced a rk3399 4gb SBC that runs the same SoC as the Pinebook Pro, can definitely get a couple of these to the right people here if that would help.  I know there's significant overlap in interest between Pinebook Pro users/developers/enthusiasts and Zephyr.

Thanks so much.

closed 2020-02-05 by evantaylor