zephyrproject-rtos/sdk-ng #182: newlib-nano not supported on all targets

galak, 2020-02-05:
Need to add spec file for nano configs that are supported by default in newlib.

galak, 2020-02-05:
@r2r0 fyi

SebastianBoe, 2020-03-31:
x86 is not supported, but ARM is, right?

SebastianBoe, 2020-03-31:
Nevermind, I answered my own question:

> sebo@sebo-nix:~/ncs/nrf/applications/asset_tracker$ find /home/sebo/zephyr-sdk-0.11.1/ | grep nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv32imafc/ilp32f/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv64imac/lp64/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv64imac/lp64/medany/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv32i/ilp32/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv64imafdc/lp64d/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv64imafdc/lp64d/medany/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv32imac/ilp32/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv32iac/ilp32/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/riscv64-zephyr-elf/riscv64-zephyr-elf/lib/rv32im/ilp32/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/arm/v5te/softfp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/arm/v5te/hard/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v8-m.main+dp/softfp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v8-m.main+dp/hard/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m+fp/softfp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m+fp/hard/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v8-m.base/nofp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7-m/nofp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v8-m.main+fp/softfp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v8-m.main+fp/hard/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v8-m.main/nofp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/nofp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7+fp/softfp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7+fp/hard/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m+dp/softfp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m+dp/hard/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v6-m/nofp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7/nofp/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/nps400/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc601_norm/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/hs38_linux/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arcem/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/em4_fpus/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc600_norm/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/em4_dmips/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc700/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/hs38/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc601/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc600_mul64/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/hs4xd/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/em4_fpuda/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/em/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/em4/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/hs/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc600/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc601_mul64/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/em_mini/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc600_mul32x16/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/quarkse_em/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/archs/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/arc601_mul32x16/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/hs34/nano.specs
> /home/sebo/zephyr-sdk-0.11.1/arc-zephyr-elf/arc-zephyr-elf/lib/hs4x/nano.specs

stephanosio, 2021-08-03:
Closing stale issue.

closed 2021-08-03 by stephanosio