zephyrproject-rtos/sdk-ng #192: Upgrade QEMU to 5.0.0

stephanosio, 2020-02-26:
Upgrade QEMU to version 5.0.0 (release expected on 28 April 2020).

Also, try to include the QEMU 5.0.0 RC versions in the Zephyr SDK 0.12.0 beta releases for testing.

https://wiki.qemu.org/ChangeLog/5.0
https://wiki.qemu.org/Planning/5.0

This is essential for https://github.com/zephyrproject-rtos/zephyr/issues/22870.

stephanosio, 2020-02-26:
cc @ioannisg @galak

stephanosio, 2020-03-01:
The following patches, if not merged before QEMU 5.0-rc0, must be added:
https://lists.gnu.org/archive/html/qemu-devel/2020-02/msg07841.html
https://lists.gnu.org/archive/html/qemu-devel/2020-02/msg07842.html
https://lists.gnu.org/archive/html/qemu-devel/2020-02/msg08453.html
https://lists.gnu.org/archive/html/qemu-devel/2020-03/msg00001.html

closed 2020-07-10 by galak

abrodkin, 2020-07-13:
@galak @stephanosio any ETAs on new SDK availability? Would be good to have whatever QEMU version with `icount` for ARC to flush out those sporadic failures we used to see and then get PRs running on ARC simulation again.

galak, 2020-07-13:
> @galak @stephanosio any ETAs on new SDK availability? Would be good to have whatever QEMU version with `icount` for ARC to flush out those sporadic failures we used to see and then get PRs running on ARC simulation again.

Did you guys implement icount support for ARC?

abrodkin, 2020-07-13:
> 
> 
> > @galak @stephanosio any ETAs on new SDK availability? Would be good to have whatever QEMU version with `icount` for ARC to flush out those sporadic failures we used to see and then get PRs running on ARC simulation again.
> 
> Did you guys implement icount support for ARC?

@galak well you're assigned as a reviewer to this one BTW https://github.com/zephyrproject-rtos/zephyr/pull/26646 :)

galak, 2020-07-13:
@abrodkin ah, too many PRs :).  Can you submit a PR to the sdk-ng with that icount change?