zephyrproject-rtos/sdk-ng #196: dependency on python 3.6 headers breaks on debian

danmcb, 2020-03-13:
/home/daniel/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb: error while loading shared libraries: libpython3.6m.so.1.0: cannot open shared object file: No such file or directory

this is because on recent debian/ubuntu 3.6 headers don't seem to be available.  See:

https://tracker.debian.org/pkg/python3.6

there are some "workarounds"  out there .. like:

https://stackoverflow.com/questions/43621584/why-cant-i-install-python3-6-dev-on-ubuntu16-04

but at least for me I still can't get the 3.6 headers (3.7 headers ARE easily installed with package python3-dev ...)

GeorgeGkinis, 2020-03-15:
Came against this issue. 
Resolved it by a dirty hack : 
```
vagrant@10:~/nRF5/ble-beacon-zephyr$ locate libpython3.7m.so.1.0
/home/linuxbrew/.linuxbrew/Cellar/python/3.7.6_1/lib/libpython3.7m.so.1.0
/home/linuxbrew/.linuxbrew/lib/libpython3.7m.so.1.0
/usr/lib/x86_64-linux-gnu/libpython3.7m.so.1.0
```
```
vagrant@10:~/nRF5/ble-beacon-zephyr$ ln -s /usr/lib/x86_64-linux-gnu/libpython3.7m.so.1.0 /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
```

pabigot, 2020-03-23:
@galak This appears to be a result of a change in 0.11.2 to dynamically link instead of statically link as in previous versions.

The hack solutions are really not an acceptable workaround.  If there isn't a critical reason why the change to dynamic linking had to be made Zephyr SDKs should use static linking for anything that isn't part of the toolchain download.

My workaround is to return to 0.11.1.

galak, 2020-03-23:
This is all related to the version of python that exists when we build the SDK.  In SDK 0.10.x timeframe we built with python2.7 as the host ubuntu system was ubuntu 16.04.  We updated to 18.04 and as such the python version bumped to 3.6.

I attempted to fix this by building a static gdb which is what sdk 0.11.2 happens to have.  However if one does that it breaks the python support.  Thus reverted back to non-static build.

pabigot, 2020-03-23:
I still don't get it, since it was a change between 0.11.1 and 0.11.2 that broke things, and 0.11.2 introduced a dynamic gdb.  Maybe 0.11.2 came after the move to 18.04, and the rest were before?

In any case, I'm going to use 0.11.1 when I need a debugger.

```
tirzah[167]$ ldd /usr/local/zephyr-sdk-0.11.?/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb
/usr/local/zephyr-sdk-0.11.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb:
	not a dynamic executable
/usr/local/zephyr-sdk-0.11.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb:
	not a dynamic executable
/usr/local/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb:
	linux-vdso.so.1 (0x00007ffefb9e4000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0c18439000)
	libpython3.6m.so.1.0 => not found
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0c18416000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f0c182c7000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f0c180d9000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f0c180bf000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0c17ecc000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f0c18db5000)
```

galak, 2020-03-23:
> I still don't get it, since it was a change between 0.11.1 and 0.11.2 that broke things, and 0.11.2 introduced a dynamic gdb. Maybe 0.11.2 came after the move to 18.04, and the rest were before?
>
> In [...]

We had dynamic GDB in 0.10.x, I broke gdb python support in 0.11.1 by trying to move to static, that was "fixed" in 0.11.2.

stephanosio, 2020-04-10:
This is not really a "Zephyr SDK" issue per se, is it?

We have always used dynamic linking for GDB and it is just that the Zephyr SDK 0.11 introduced a new Python version dependency (i.e. 3.6 instead of 2.7 like it used to). Installing all the required dependencies, as we do for the others, should make it work.

Moreover, Ubuntu 16.04 LTS has reached end-of-life; so you should consider upgrading to 18.04 or 20.04.

stephanosio, 2020-04-10:
I have submitted a PR to force dynamic-linking the GDB against `libpython2.7` as we used to do (see https://github.com/zephyrproject-rtos/sdk-ng/pull/202).

That PR should resolve this issue.

stephanosio, 2020-04-10:
Also note that GDB should, in theory, support static-linking of `libpython`.

The reason that we saw Python scripting capability being disabled when we enabled static linking (`CT_GDB_CROSS_STATIC=y`) is that [`GDB_CROSS_PYTHON` depends on `!GDB_CROSS_STATIC`](https://github.com/crosstool-ng/crosstool-ng/blob/d5900debd397b8909d9cafeb9a1093fb7a5dc6e6/config/debug/gdb.in.cross#L41) and [`CT_GDB_CROSS_PYTHON=n` resolves to `--with-python=no`](https://github.com/crosstool-ng/crosstool-ng/blob/75d7525aa1698a8bd8dd0c6b78819846fe2ec869/scripts/build/debug/300-gdb.sh#L40-L41).

pabigot, 2020-04-10:
> Moreover, Ubuntu 16.04 LTS has reached end-of-life; so you should consider upgrading to 18.04 or 20.04.

[LTS](https://wiki.ubuntu.com/Releases) support in Ubuntu is five years, so it's still supported until April 2021, and doesn't EOL (security patches stop) until April 2024.

That pedantry aside, thank you for providing a fix for this.  I look forward to 0.11.3.

stephanosio, 2020-04-10:
> [LTS](https://wiki.ubuntu.com/Releases) support in Ubuntu is five years, so it's still supported until April 2021, and doesn't EOL (security patches stop) until April 2024.

ah, I got this mixed up with something else.

galak, 2020-04-10:
> Also note that GDB should, in theory, support static-linking of `libpython`.
>
> The reason that we saw Python scripting capability being disabled when we enabled static linking [...]

can we look to fix crosstool.  I'd prefer we link statically with python3.6 if we can.

stephanosio, 2020-04-10:
> can we look to fix crosstool. I'd prefer we link statically with python3.6 if we can.

I wonder if that should be a long-term goal, as doing so might have other side effects and,  at least according to crosstool, [there have been reports of problems when linking gdb to the static libpython.a in older versions](https://github.com/crosstool-ng/crosstool-ng/commit/c7a396f9b7f61bb6927b276188c075e47e9db0e7#diff-8c164ab2921b8723a2ae7193629fd8beR48-R50).

galak, 2020-04-10:
> I wonder if that should be a long-term goal, as doing so might have other side effects and, at least according to crosstool, [there have been reports of problems when linking gdb to the static [...]

saw that comment and I took that to mean issues when building.  So I figure we should remove the `depends on ! GDB_CROSS_STATIC` and see what happens.

andrewboie, 2020-05-08:
I wonder if it's feasible to provide a "snap" of the Zephyr SDK, which will eliminate dependency problems as everything is contained in the image.

https://snapcraft.io/

closed 2020-05-15 by galak

sylvainpelissier, 2021-09-22:
I have now the same problem with Ubuntu 21.04 where python 3.8 is not available anymore:
```
/home/user/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
```

stephanosio, 2021-09-22:
> I have now the same problem with Ubuntu 21.04 where python 3.8 is not available anymore:

@sylvainpelissier We have added the `*-gdb-no-py` variant (e.g. `arm-zephyr-eabi-gdb-no-py`) with this in mind.

If you must use the Python scripting functionality in the GDB, please install Python 3.8 on your system.

sylvainpelissier, 2021-09-22:
I have Python 3.9.5 but 3.8 is not available anymore on Ubuntu 21.04 for installation.

stephanosio, 2021-09-22:
> I have Python 3.9.5 but 3.8 is not available anymore on Ubuntu 21.04 for installation.

@sylvainpelissier You can install third-party packages (you should be able to find a few if you google "ubuntu 21.04 python 3.8"), or build and install Python 3.8 from source.

carlescufi, 2021-09-22:
See https://github.com/zephyrproject-rtos/zephyr/pull/38749