zephyrproject-rtos/sdk-ng #198: How to build latest toolchain for riscv and add/update it in the SDK?

musamaanjum, 2020-03-30:
I'm using zephyr with its eclipse plugin. I just want to replace the latest RISC-V toolchain in the SDK for research purposes.

I'm trying to build riscv64 and riscv32 toolchain from https://github.com/riscv/riscv-gnu-toolchain (with program prefix riscv64-zephyr-elf). When I compare my own built toolchain with the toolchain present in the zephyr SDK, there are many differences. The most prominent difference is the ld scripts present in _riscv64-zephyr-elf/lib/ldscripts_. [comparison.txt](https://github.com/zephyrproject-rtos/sdk-ng/files/4405165/comparison.txt) is here.

So my question is that how can I build the latest toolchain from (https://github.com/riscv/riscv-gnu-toolchain) and put that in the SDK? What changes I should do after building it so that my built toolchain works exactly the toolchain present in the SDK works?

galak, 2020-03-30:
Why do you need to put the toolchain build from riscv-gnu-toolchain into the SDK?  Why can't you just configure your zephyr env to utilize that toolchain?

galak, 2020-03-30:
For example:

https://docs.zephyrproject.org/latest/getting_started/toolchain_other_x_compilers.html?highlight=cross%20compiler

musamaanjum, 2020-04-04:
Yeah, setting the cross compiler is way easy. Thanks.

closed 2020-04-04 by musamaanjum