zephyrproject-rtos/sdk-ng #211: v0.11.30-beta-2 assets still look very much like the ones from beta-1

abrodkin, 2020-05-12:
https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.11.3-beta-2

See: https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.11.3-beta-2/zephyr-sdk-0.11.3-beta-1-setup.run as well as console messages during installation also mention `beta1`:
```
./zephyr-sdk-0.11.3-beta-1-setup.run -- -d .../username/Tools/zephyr-sdk/0.11.3-beta-2/
Verifying archive integrity... All good.
Uncompressing SDK for Zephyr  100%

It is recommended to install Zephyr SDK at one of the following locations:
  /u/username/zephyr-sdk[-0.11.3-beta-1]
  /u/username/.local/zephyr-sdk[-0.11.3-beta-1]
  /u/username/.local/opt/zephyr-sdk[-0.11.3-beta-1]
  /u/username/bin/zephyr-sdk[-0.11.3-beta-1]
  /opt/zephyr-sdk[-0.11.3-beta-1]
  /usr/zephyr-sdk[-0.11.3-beta-1]
  /usr/local/zephyr-sdk[-0.11.3-beta-1]
```

Even though internals look like being really `beta2` as there's QEMU for ARC :)

So that's just something wrong with naming.

stephanosio, 2020-05-12:
Forgot to update `VERSION` file:
https://github.com/zephyrproject-rtos/sdk-ng/blob/master/VERSION

galak, 2020-05-12:
Yeah, I forgot to update it.  Will fix when I make the actual 0.11.3 release or next beta.

closed 2020-05-12 by galak