zephyrproject-rtos/sdk-ng #225: Big-Endian toolchain for Cortex-R/A

paoloteti, 2020-06-05:
During last weekend I decided to give a try to Zephyr on some Cortex-R board.
First issue I found was that Zephyr SDK doesn't contain a Big-Endian toolchain (lib-nano is LE only) and I own only BE Cortex-R boards from Texas Instruments.
This was not a problem at all because I easily switched to `armeb-none-eabi` toolchain I use for bare-metal projects.
I'm wondering why there isn't an `armeb-zephyr-elf` toolchain targetting Cortex-R/A BE variant and if there is a plan to add it in the future.

stephanosio, 2020-06-05:
https://github.com/stephanosio/zephyr-crosstool-ng/issues/64

stephanosio, 2020-06-05:
Also note that the ARM Cortex-R arch port has never been tested on big endian platforms, so there will likely be a quite a few things that do not work (if it boots at all, that is).

stephanosio, 2021-08-03:
Closing for now as there currently is no SoC/board on the main Zephyr tree that requires the big-endian ARM toolchain.

This issue can be re-opened when there is an actual demand for the big-endian ARM toolchain in the upstream Zephyr repository.

closed 2021-08-03 by stephanosio

ntavish, 2024-05-27:
Hello, I would like to re-open this issue. Me, and seems like a couple other people may have TI Hercules boards that we may want to add to upstream zephyr. I am starting to work on it now.