zephyrproject-rtos/sdk-ng #226: Build fails with '_PyImport_FixupBuiltin' too few arguments error

finikorg, 2020-06-08:
Building some PR fails, for example when building PR #193 I have following error:
...
ERROR]    sdk-ng/build/build_xtensa_intel_s1000/.build/xtensa-zephyr-elf/src/gdb/gdb/python/python.c:1577:3: error: too few arguments to function '_PyImport_FixupBuiltin'
[ERROR]    make[3]: *** [Makefile:2057: python.o] Error 1
[ERROR]    make[3]: *** Waiting for unfinished jobs....
[ERROR]    make[2]: *** [Makefile:8258: all-gdb] Error 2
[ERROR]    make[1]: *** [Makefile:828: all] Error 2
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@376]
[ERROR]  >>        called from: do_gdb_backend[scripts/build/debug/300-gdb.sh@281]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@59]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@696]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      https://crosstool-ng.github.io/docs/known-issues/
[ERROR]  >>
[ERROR]  >> NOTE: Your configuration includes features marked EXPERIMENTAL.
[ERROR]  >> Before submitting a bug report, try to reproduce it without enabling
[ERROR]  >> any experimental features. Otherwise, you'll need to debug it
[ERROR]  >> and present an explanation why it is a bug in crosstool-NG - or
[ERROR]  >> preferably, a fix.
[ERROR]  >>
[ERROR]  >> NOTE: You configuration uses non-default patch sets. Please
[ERROR]  >> select 'bundled' as the set of patches applied and attempt
[ERROR]  >> to reproduce this issue. Issues reported with other patch
[ERROR]  >> set selections (none, local, bundled+local) are going to be
[ERROR]  >> closed without explanation.
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]
[ERROR]  (elapsed: 21:25.62)
make: *** [sdk-ng/bin/ct-ng:261: build] Error 2
...

stephanosio, 2020-06-08:
I suppose you get this error when building locally. If so, please ensure the following:

1. Check out the latest copy of the sdk-ng repository
2. Build GDB 8.3 for `xtensa_intel_s1000` (`build.log` should give you a rough idea)
3. Python 3.8 development package installed (in case of Ubuntu 18.04, install `python3.8-dev`)
4. Python 3.8 executable can be globally resolved as `python3.8` (make sure that `python3.8 --version` actually prints `Python 3.8.x`).

finikorg, 2020-06-15:
Closing issues, after I started use newer GDB version.

closed 2020-06-15 by finikorg