zephyrproject-rtos/sdk-ng #227: DTC version doesn't work

Kreijstal, 2020-06-10:
Install sdk anywhere, and try to run dtc.

`zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: error while loading shared libraries: libunwind.so.8: cannot open shared object file: No such file or directory`

stephanosio, 2020-06-10:
What distro are you using?

Kreijstal, 2020-06-10:
@stephanosio Ubuntu 18.04.3 LTS

stephanosio, 2020-06-10:
Doing `sudo apt install libunwind8` should fix the problem.

Kreijstal, 2020-06-10:
@stephanosio it doesn't

stephanosio, 2020-06-10:
Do you have `/usr/lib/x86_64-linux-gnu/libunwind.so.8`?

Kreijstal, 2020-06-10:
@stephanosio it seems so check this out https://colab.research.google.com/drive/1oFDmZDziioQtZHlc0QBgEWTemJyYVQxV

stephanosio, 2020-06-10:
Please post the result of the following commands here:
```
```

```
/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
```

Kreijstal, 2020-06-10:
1st
```
lrwxrwxrwx 1 root root 18 Apr 12  2018 /usr/lib/x86_64-linux-gnu/libunwind.so.8 -> libunwind.so.8.0.1
```
2nd
```
	linux-vdso.so.1 (0x00007ffd83125000)
	/usr/lib/x86_64-linux-gnu/libtcmalloc.so.4 (0x00007f06459e9000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f06455f8000)
	libunwind.so.8 => /usr/lib/x86_64-linux-gnu/libunwind.so.8 (0x00007f06453dd000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f06451be000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f0644e35000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f0644a97000)
	/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2 (0x00007f0645c59000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f0644871000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f0644659000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0644455000)
```

stephanosio, 2020-06-10:
So `libunwind.so.8` is clearly there and the loader is properly resolving it, yet you are getting "libunwind.so.8: cannot open shared object file: No such file or directory."

I think there might be something seriously wrong with your system.

stephanosio, 2020-06-10:
Just in case, see if this fixes the problem:
```
sudo patchelf --set-interpreter /lib64/ld-linux-x86-64.so.2 /usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
```

Kreijstal, 2020-06-10:
I just straced it and this is what I get out of it
```
execve("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc", ["/usr/zephyr-sdk-0.11.3/sysroots/"...], 0x7fffaefbfa30 /* 41 vars */) = 0
brk(NULL)                               = 0x17e6000
mmap(NULL, 331776, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f207f246000
open("/usr/lib/x86_64-linux-gnu/libtcmalloc.so.4", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\360(\1\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=290632, ...}) = 0
mmap(NULL, 2553952, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f207ee03000
mprotect(0x7f207ee48000, 2097152, PROT_NONE) = 0
mmap(0x7f207f048000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x45000) = 0x7f207f048000
mmap(0x7f207f04a000, 165984, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f207f04a000
close(3)                                = 0
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/usr/local/nvidia/lib/tls/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/nvidia/lib/tls/x86_64", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/local/nvidia/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/nvidia/lib/tls", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/local/nvidia/lib/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/nvidia/lib/x86_64", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/local/nvidia/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/nvidia/lib", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/local/nvidia/lib64/tls/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/nvidia/lib64/tls/x86_64", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/local/nvidia/lib64/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/nvidia/lib64/tls", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/local/nvidia/lib64/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/nvidia/lib64/x86_64", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/local/nvidia/lib64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/local/nvidia/lib64", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/tls/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/tls/x86_64", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/tls", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/x86_64", 0x7fffa4579bc0) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260\3\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1697520, ...}) = 0
mmap(NULL, 3803552, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f207ea62000
mprotect(0x7f207ebf7000, 2097152, PROT_NONE) = 0
mmap(0x7f207edf7000, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x195000) = 0x7f207edf7000
mmap(0x7f207edff000, 14752, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f207edff000
close(3)                                = 0
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/libunwind.so.8", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib/tls/x86_64/libunwind.so.8", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib/tls/x86_64", 0x7fffa4579a70) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib/tls/libunwind.so.8", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib/tls", 0x7fffa4579a70) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib/x86_64/libunwind.so.8", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib/x86_64", 0x7fffa4579a70) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib/libunwind.so.8", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
[... cut ...] _len=1}], 10/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: error while loading shared libraries: libunwind.so.8: cannot open shared object file: No such file or directory
) = 178
exit_group(127)                         = ?
+++ exited with 127 +++
```
so basically it doesn't actually look for `/usr/lib/x86_64-linux-gnu/libunwind.so.8` it looks for `/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/lib/libunwind.so.8` which isn't actually there, so it gives file not found

Kreijstal, 2020-06-10:
> Just in case, see if this fixes the problem:
> 
> ```
> sudo patchelf --set-interpreter /lib64/ld-linux-x86-64.so.2 /usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
> ```

well, it does fix the problem, but uh, I shouldn't need to do this.

stephanosio, 2020-06-10:
> well, it does fix the problem, but uh, I shouldn't need to do this.

Normally, it would load the host system lib if `/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2` cannot find one in the Poky sysroot; but for some reason, it is not doing that on your system.

And yes, that library should be, strictly speaking, included in the Poky sysroot.

stephanosio, 2020-06-10:
@Kreijstal I just checked my local SDK installation and I am not getting any of those dependencies:
```
$ ldd /opt/sdk/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
        linux-vdso.so.1 (0x00007ffe8cf68000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f413bc2d000)
        /opt/sdk/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2 (0x00007f413c01e000)
```

Did you do something to the SDK?

Kreijstal, 2020-06-10:
> @Kreijstal I just checked my local SDK installation and I am not getting any of those dependencies:
>
> ```
> $ ldd /opt/sdk/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc [...]

I'm testing it in google colab, so it's a bare-bones system. All I've ever done was install the sdk, and get into the error.

stephanosio, 2020-06-10:
> I'm testing it in google colab, so it's a bare-bones system. All I've ever done was install the sdk, and get into the error.

I am baffled by what you get from `ldd`-ing on the Zephyr SDK `dtc` executable (`ldd /usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc`).

It should be:
```
        linux-vdso.so.1 (0x00007ffe8cf68000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f413bc2d000)
        /opt/sdk/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2 (0x00007f413c01e000)
```

Not:
```
	linux-vdso.so.1 (0x00007ffd83125000)
	/usr/lib/x86_64-linux-gnu/libtcmalloc.so.4 (0x00007f06459e9000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f06455f8000)
	libunwind.so.8 => /usr/lib/x86_64-linux-gnu/libunwind.so.8 (0x00007f06453dd000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f06451be000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f0644e35000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f0644a97000)
	/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2 (0x00007f0645c59000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f0644871000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f0644659000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0644455000)
```

As far as I can tell, this is not a problem with the Zephyr SDK.

stephanosio, 2020-06-10:
@Kreijstal What do you get when you do:
```
ldd -v /usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc
```

Kreijstal, 2020-06-10:
> ldd -v /usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc

```
linux-vdso.so.1 (0x00007ffd136d1000)
	/usr/lib/x86_64-linux-gnu/libtcmalloc.so.4 (0x00007f132fb13000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f132f722000)
	libunwind.so.8 => /usr/lib/x86_64-linux-gnu/libunwind.so.8 (0x00007f132f507000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f132f2e8000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f132ef5f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f132ebc1000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f132fd83000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f132e99b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f132e783000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f132e57f000)

	Version information:
	/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc:
		libc.so.6 (GLIBC_2.3) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
	/usr/lib/x86_64-linux-gnu/libtcmalloc.so.4:
		libm.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libm.so.6
		libstdc++.so.6 (CXXABI_1.3.8) => /usr/lib/x86_64-linux-gnu/libstdc++.so.6
		libstdc++.so.6 (CXXABI_1.3) => /usr/lib/x86_64-linux-gnu/libstdc++.so.6
		libstdc++.so.6 (GLIBCXX_3.4.21) => /usr/lib/x86_64-linux-gnu/libstdc++.so.6
		libstdc++.so.6 (GLIBCXX_3.4) => /usr/lib/x86_64-linux-gnu/libstdc++.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.27) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
		libpthread.so.0 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libpthread.so.0
	/lib/x86_64-linux-gnu/libc.so.6:
		ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
		ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2
	/usr/lib/x86_64-linux-gnu/libunwind.so.8:
		ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
		liblzma.so.5 (XZ_5.0) => /lib/x86_64-linux-gnu/liblzma.so.5
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libpthread.so.0:
		ld-linux-x86-64.so.2 (GLIBC_2.2.5) => /lib64/ld-linux-x86-64.so.2
		ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.2) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_PRIVATE) => /lib/x86_64-linux-gnu/libc.so.6
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6:
		libm.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libm.so.6
		ld-linux-x86-64.so.2 (GLIBC_2.3) => /lib64/ld-linux-x86-64.so.2
		libgcc_s.so.1 (GCC_4.2.0) => /lib/x86_64-linux-gnu/libgcc_s.so.1
		libgcc_s.so.1 (GCC_3.3) => /lib/x86_64-linux-gnu/libgcc_s.so.1
		libgcc_s.so.1 (GCC_3.0) => /lib/x86_64-linux-gnu/libgcc_s.so.1
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.18) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.16) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.17) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.3.2) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libm.so.6:
		ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_PRIVATE) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/liblzma.so.5:
		libdl.so.2 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libdl.so.2
		libpthread.so.0 (GLIBC_2.3.3) => /lib/x86_64-linux-gnu/libpthread.so.0
		libpthread.so.0 (GLIBC_2.3.2) => /lib/x86_64-linux-gnu/libpthread.so.0
		libpthread.so.0 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libpthread.so.0
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.17) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libgcc_s.so.1:
		libc.so.6 (GLIBC_2.14) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
	/lib/x86_64-linux-gnu/libdl.so.2:
		ld-linux-x86-64.so.2 (GLIBC_PRIVATE) => /lib64/ld-linux-x86-64.so.2
		libc.so.6 (GLIBC_PRIVATE) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.4) => /lib/x86_64-linux-gnu/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /lib/x86_64-linux-gnu/libc.so.6
```
It might be because this is containarized somehow, but not sure.

stephanosio, 2020-06-10:
I think Google Colab is doing some magickery in their dynamic linker.

This is what strace should look like:
```
execve("./dtc", ["./dtc"], 0x7fffeac19b70 /* 37 vars */) = 0
brk(NULL)                               = 0x7b4000
mmap(NULL, 331776, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f48ca0a3000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/tls/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/tls/x86_64", 0x7ffc8edde9d0) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/tls", 0x7ffc8edde9d0) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/x86_64/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/x86_64", 0x7ffc8edde9d0) = -1 ENOENT (No such file or directory)
open("/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260\3\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1697520, ...}) = 0
mmap(NULL, 3803552, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f48c9b2f000
mprotect(0x7f48c9cc4000, 2097152, PROT_NONE) = 0
mmap(0x7f48c9ec4000, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x195000) = 0x7f48c9ec4000
mmap(0x7f48c9ecc000, 14752, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f48c9ecc000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f48ca0a1000
arch_prctl(ARCH_SET_FS, 0x7f48ca0a1700) = 0
mprotect(0x7f48c9ec4000, 16384, PROT_READ) = 0
mprotect(0x618000, 4096, PROT_READ)     = 0
mprotect(0x7f48ca0f4000, 4096, PROT_READ) = 0
stat("-", 0x7ffc8eddf1e0)               = -1 ENOENT (No such file or directory)
brk(NULL)                               = 0x7b4000
brk(0x7d5000)                           = 0x7d5000
fstat(0, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 80), ...}) = 0
read(0,
```

As far as Zephyr SDK is concerned, the `dtc` executable only depends on `libc.so.6`.

```
/usr/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin$ readelf -d ./dtc

Dynamic section at offset 0x18de8 contains 24 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000c (INIT)               0x402320
 0x000000000000000d (FINI)               0x40ec90
 0x0000000000000019 (INIT_ARRAY)         0x618728
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0x618730
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x0000000000000004 (HASH)               0x401280
 0x0000000000000005 (STRTAB)             0x401b00
 0x0000000000000006 (SYMTAB)             0x401548
 0x000000000000000a (STRSZ)              469 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000015 (DEBUG)              0x0
 0x0000000000000003 (PLTGOT)             0x619000
 0x0000000000000002 (PLTRELSZ)           1272 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x401e28
 0x0000000000000007 (RELA)               0x401d80
 0x0000000000000008 (RELASZ)             168 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffe (VERNEED)            0x401d50
 0x000000006fffffff (VERNEEDNUM)         1
 0x000000006ffffff0 (VERSYM)             0x401cd6
 0x0000000000000000 (NULL)               0x0
```

Kreijstal, 2020-06-10:
Very Interesting.

stephanosio, 2021-08-03:
Closing stale issue.

closed 2021-08-03 by stephanosio