zephyrproject-rtos/sdk-ng #234: Update BOSSA to 1.9.1

nzmichaelh, 2020-06-28:
b20b0aaed93dd5a074d435e3ee6af4ffb4b4434b was the previous attempt.  Rolling forward requires changes to the Zephyr side west command.  Once that has landed, this bug will track updating the SDK.

stephanosio, 2020-06-29:
@galak I opened a PR that brings back the old commit: https://github.com/zephyrproject-rtos/sdk-ng/pull/235

There should be no problem upgrading to the latest version of BOSSA since the offset param is now handled by https://github.com/zephyrproject-rtos/zephyr/pull/26510.

nzmichaelh, 2020-06-29:
Could we pick up shumatech/BOSSA/pull/78 as well?  It fixes a re-enumeration issue when resetting.

nzmichaelh, 2020-06-30:
Oh, and this will add support for Windows machines as we can drop the stty reset trigger and use BOSSA's built-in reset support.

galak, 2020-07-10:
@nzmichaelh is there a fix for https://github.com/shumatech/BOSSA/issues/96 ?  That was why we went back to 1.7.

nzmichaelh, 2020-07-10:
Hi @galak I'm not sure the one you bisected to is the cause or, at least, the only one.

schumatech/BOSSA@dbdd088909c57bdf9aee191720bf6c6159217e22 made a breaking change.  In 1.7 and 1.8 it would automatically detect the bootloader and add the offset, while in 1.9 you need to explicitly supply the `--offset`.  The PR that I sent recently to runners/bossa.py now detects the flag and adds it automatically based on the DeviceTree partitions.

I've tested on my machine with the Ubuntu supplied 1.9.1 and it works fine.  Any thoughts on how else we can test?

nzmichaelh, 2020-07-10:
Also, if we go ahead, could we pick up the current HEAD schumatech/BOSSA@3532de82efd28fadbabc2b258d84dddf14298107 ?  It fixes a race in re-enumeration which we'll otherwise have to work-around in runners/bossa.py.

galak, 2020-07-10:
I think there was some issue flashing the Arduino 101 if I remember correctly.  @nashif might also vaguely remember this.

nashif, 2020-07-10:
> I think there was some issue flashing the Arduino 101 if I remember correctly. @nashif might also vaguely remember this.

not 101, this was about Arduino Due I think :)

galak, 2020-07-13:
@nzmichaelh do you know what the --offset should be for arduino_due.  Trying to test that out w/top of tree BOSSA to make sure that works ok.

galak, 2020-07-13:
@nzmichaelh a few different things:

1. looks like main line of BOSSA has changed exits code for --help to return 0, so we need west runner to handle that (https://github.com/shumatech/BOSSA/commit/f83f741d07ef2d4fb10404a74bf2b9a9aa01ec09)
2. When I try to run with bossa 1.9.1 and flash arduino_due I get:
```
/home/galak/git/BOSSA/bin/bossac -p /dev/ttyACM0 -R -e -w -v -b /home/galak/git/zephyr/samples/hello_world/build/zephyr/zephyr.bin -o 524288
Erase flash

Done in 0.000 seconds

Flash offset is invalid
```

nzmichaelh, 2020-07-13:
re: exit code, I'll fix this.

re: offset, I read up on the Due and it has a separate ATMEGA as an onboard programmer, and the flash is at offset 0x80000 which matches the `-o 524288`.  I'll investigate further.

Was this against an actual Due, or some other Arduino?  I ask as if BOSSA detects the memory, then +512k would truly be invalid as most SAMDs have 256k of flash from address 0.

nzmichaelh, 2020-07-13:
(and you can see what BOSSA detected by adding a `-i` to the command line)

galak, 2020-07-13:
> re: exit code, I'll fix this.
>
> re: offset, I read up on the Due and it has a separate ATMEGA as an onboard programmer, and the flash is at offset 0x80000 which matches the `-o 524288`. I'll [...]

This was against an actual Due board.

```
[galak@lava2 ~]$ /home/galak/git/BOSSA/bin/bossac -p /dev/ttyACM0 -i
Device       : ATSAM3X8
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 0x80000
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : none
Security     : false
Boot Flash   : false
```

nzmichaelh, 2020-07-14:
Thanks.  `--offset` might be the offset into the flash instead of the offset into the address space.  I'll investigate and send two PRs:

- Fix the `--help` processing on the git version of BOSSA
- Investigate `--offset` and update to offset-in-flash

Only the SAM3X is affected by this.  All other BOSSA based boards in Zephyr have flash that start at zero

galak, 2020-07-16:
Handled by commit e097bbdd89f0333e00f27d6322836b2aa5bb413a

closed 2020-07-16 by galak