zephyrproject-rtos/sdk-ng #236: Set sysroot in gcc builds

keith-packard, 2020-07-07:
I'm working on integrating picolibc into zephyr-sdk and need to refer to directories relative to the zephyr sdk install location. By defining sysroot, the picolibc .specs file can use %R to reference locations relative to the SDK rather than needing absolute paths.

The existing nano.specs files are already assuming a sysroot definition as they use -isystem =/include/newlib-nano which currently fails to do the right thing:

$ arm-zephyr-eabi-gcc -v -S --specs=nano.specs foo.c
...
ignoring nonexistent directory "=/include/newlib-nano"
...

galak, 2020-07-08:
Hmm, seems like crosstool-ng has sysroot disabled for bare metal builds:

https://github.com/crosstool-ng/crosstool-ng/commit/01bae4871994429725a34298787a04221fd1368d

keith-packard, 2020-07-08:
> Hmm, seems like crosstool-ng has sysroot disabled for bare metal builds:
> 
> [crosstool-ng/crosstool-ng@01bae48](https://github.com/crosstool-ng/crosstool-ng/commit/01bae4871994429725a34298787a04221fd1368d)

The comment there seems incorrect to me -- sysroot has no relation to the target machine..

stephanosio, 2021-08-03:
Closing stale issue.

IIUC, this is probably no longer required since `GCC_EXEC_PREFIX` can serve the same purpose:
https://github.com/crosstool-ng/crosstool-ng/commit/89273b59af94fa1a40991762c780f668a7f032ec#diff-e8617de7f25b204d36f49a3ddd5fad51da05aaf4737789b517bcf5bbca59450b

closed 2021-08-03 by stephanosio