zephyrproject-rtos/sdk-ng #255: stuck qemu-system-aarch64 processes from sanitycheck

andrewboie, 2020-07-21:
**Describe the bug**
I have noticed that when running sanitycheck, `qemu-system-aarch64` processes will dangle consuming 100% CPU if I ctrl-C out of a sanitycheck run. They ignore SIGTERM, I have to send SIGKILL to terminate them.

I also notice when running tests manually on qemu_cortex_a53 that the test will complete but I cannot Ctrl-C out of the emulator, unlike our other QEMU builds. Something wrong with our SDK build?

**To Reproduce**
Fire up sanitycheck -p qemu_cortex_a53 and ctrl-c out of sanitycheck while it is doing stuff. Notice stuck QEMU processes.

**Expected behavior**
Emulators to exit properly like other QEMU builds.

**Impact**
Very annoying.

galak, 2020-07-21:
will be curious to see if the update to qemu 5.0 still shows this or not.

Any way to try and reproduce this consistency?

galak, 2020-07-21:
@andrewboie - seems like this might be related to `-icount` support.  If I remove `sleep=off`.

galak, 2020-07-22:
Was able to reproduce this with just running helloworld.  Seems like there is a mutex locking issue when we run with our `-icount` args.  I've asked one of the qemu maintainers to take a look.

galak, 2020-07-28:
https://patchew.org/QEMU/20200728141005.28664-1-alex.bennee@linaro.org/

stephanosio, 2021-08-03:
> I've asked one of the qemu maintainers to take a look.

@galak Has there been any updates regarding this?

> https://patchew.org/QEMU/20200728141005.28664-1-alex.bennee@linaro.org/

I see that those patches have not been merged upstream. Assuming these patches fix the problem, should we just add them locally?

stephanosio, 2021-10-03:
@galak ping

stephanosio, 2022-06-18:
Closing since this does not seem to occur anymore as of the Zephyr SDK 0.14.2 release.

Please re-open if you think this is still an issue.

closed 2022-06-18 by stephanosio