zephyrproject-rtos/sdk-ng #269: SDK needs to be built and packaged as portage package

jackrosenthal, 2020-09-14:
Hi,

We are looking at integrating your SDK into our project for building Zephyr. We use Gentoo's portage for our build system.

But there are a few issues with making a portage package:

Portage expects that all sources needed are listed in `SRC_URI`, and that once the package is building, there will be no network access.

I was looking thru the "go" script, and it looks to be this relies on calling `git clone` on random GitHub repos many times.

Is there a way we can clone the sources ahead of time, and pass the appropriate repos as arguments to your scripts?

jackrosenthal, 2020-09-14:
@MaureenHelm @nashif @tejlmand @abrodkin

stephanosio, 2021-08-03:
> once the package is building, there will be no network access.

Not really a Zephyr SDK problem. You could probably look into writing a script to pre-fetch the required sources prior to building the package.

Maybe https://docs.yoctoproject.org/bitbake/bitbake-user-manual/bitbake-user-manual-fetching.html can be of use in achieving that.

closed 2021-08-03 by stephanosio