zephyrproject-rtos/sdk-ng #280: SDK 0.12.0-alpha-1 issues on CentOS

evgeniy-paltsev, 2020-10-12:
When testing `0.12.0-alpha-1` SDK I faced with issues due to raised requirements for host GLIBC version:

```
/home/arcoss_verification/tools/zephyr-sdk/0.12.0-alpha-1/arc-zephyr-elf/bin/../lib/gcc/arc-zephyr-elf/10.2.0/../../../../arc-zephyr-elf/bin/ld: /lib64/libc.so.6: version `GLIBC_2.27' not found (required by /home/arcoss_verification/tools/zephyr-sdk/0.12.0-alpha-1/arc-zephyr-elf/bin/../lib/gcc/arc-zephyr-elf/10.2.0/../../../../arc-zephyr-elf/bin/ld)
```

That make 0.12* SDK not usable on CentOS 7 (which is still supported and used).
`0.12.0-alpha-1` SDK require GLIBC 2.27 and CentOS 7 provides only 2.17:
```
ldd --version
ldd (GNU libc) 2.17
```

@galak Is it possible to soften requirements for GLIBC version in 0.12* SDK?

evgeniy-paltsev, 2020-10-12:
@abrodkin @ruuddw

galak, 2020-10-12:
Did SDK 0.11.4 work on Centos 7?

evgeniy-paltsev, 2020-10-12:
Yes, 0.11.4 works perfectly fine on Centos 7

galak, 2020-10-12:
> Yes, 0.11.4 works perfectly fine on Centos 7

I'm fine w/taking a fix to make this work, but I dont have any plans to address the issue.

abrodkin, 2020-10-12:
> > Yes, 0.11.4 works perfectly fine on Centos 7
> 
> I'm fine w/taking a fix to make this work, but I dont have any plans to address the issue.

I guess that dependency on glibc 2.27 comes from Ubuntu 18.04 which is now used in the container where SDK is built (https://github.com/zephyrproject-rtos/docker-sdk-build/blob/master/Dockerfile#L1). In that case downgrading of glibc doesn't seem to be an easy option.

I'm wondering how SDK was built previously (like 0.11.3, 0.11.4)? Was it a Poky something container?

galak, 2020-10-12:
The 0.11.3/0.11.4 are built in the same 18.04 container.  I'm guessing this is more due to binutils using something that comes from the newer glibc.  (looks like use of 'glob')

abrodkin, 2020-10-14:
@galak so that change to CT-NG allows to build Binutils w/o glob at all and `ld` built in Ubuntu 18.04 is now usable in CentOS 7.x:
```diff
diff --git a/scripts/build/binutils/binutils.sh b/scripts/build/binutils/binutils.sh
index a79bb7e2..428a3414 100644
--- a/scripts/build/binutils/binutils.sh
+++ b/scripts/build/binutils/binutils.sh
@@ -177,6 +177,10 @@ do_binutils_backend() {

     [ "${CT_TOOLCHAIN_ENABLE_NLS}" != "y" ] && extra_config+=("--disable-nls")

+    # Disable usage of glob for higher compatibility.
+    # Not strictly needed for anything but GDB anyways.
+    export ac_cv_func_glob=no
+
     CT_DoLog DEBUG "Extra config passed: '${extra_config[*]}'"

     CT_DoExecLog CFG
```

Looks reasonable? If so I'll send it as a PR to the upstream project and then pull it in here.

galak, 2020-10-14:
> Looks reasonable? If so I'll send it as a PR to the upstream project and then pull it in here.

Looks reasonable to me.

abrodkin, 2020-10-14:
Ok, @galak here we are https://github.com/crosstool-ng/crosstool-ng/pull/1408

abrodkin, 2020-10-19:
@galak now with my change accepted in the upstream CT-NG, care to cherry-pick it and rebuild the SDK?

galak, 2020-10-19:
> @galak now with my change accepted in the upstream CT-NG, care to cherry-pick it and rebuild the SDK?

Yep, first step is here:

https://github.com/zephyrproject-rtos/crosstool-ng/pull/5

closed 2020-10-19 by galak

galak, 2020-10-19:
Thanks for the fix upstream.

abrodkin, 2020-10-22:
Confirm that with https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.12.0-beta-1 the problem is solved and entire `sanitycheck` could be successfully run on CentOS 7.x.

PR #282 (2020-10-19): crosstool-ng: Update to newer upstream for centos build fix
Pull in fix for compat on centOS.  See commit in upstream crosstool-ng:

commit 57f59092852dff18fbda68fdbf23f850ad182c40
Author: Alexey Brodkin <abrodkin@synopsys.com>
Date:   Wed Oct 14 22:35:03 2020 +0300

    binutils: Disable glob for better portability

Fixes: #280
commit 339a112a (2020-10-19): crosstool-ng: Update to newer upstream for centos build fix

Pull in fix for compat on centOS.  See commit in upstream crosstool-ng:

commit 57f59092852dff18fbda68fdbf23f850ad182c40
Author: Alexey Brodkin <abrodkin@synopsys.com>
Date:   Wed Oct 14 22:35:03 2020 +0300

    binutils: Disable glob for better portability

Fixes: #280