zephyrproject-rtos/sdk-ng #285: ARC compiler does not support thread local storage

dcpleung, 2020-10-26:
When enabling thread local storage on ARC targets (`qemu_arc_em` and `qemu_arc_hs`) with `-mtls-model=local-exec`, GCC (9.2.0) emits the following error:
```
internal compiler error: in arc_legitimize_tls_address, at config/arc/arc.c:5934
```
where the code at that location is:
```
 gcc_assert (arc_tp_regno != -1);
```

This seems to caused by the code in `gcc/config/arc/elf.h`:
```
/* Bare-metal toolchains do not need a thread pointer register.  */
#undef TARGET_ARC_TP_REGNO_DEFAULT
#define TARGET_ARC_TP_REGNO_DEFAULT -1
```
but in `gcc/config/arc/linux.h`:
```
/* Linux toolchains use r25 as the thread pointer register.  */
#undef TARGET_ARC_TP_REGNO_DEFAULT
#define TARGET_ARC_TP_REGNO_DEFAULT 25
```

A correct register number needs to be defined for bare metal target on ARC before thread local storage can be enabled.

dcpleung, 2020-10-26:
@ruuddw Do you know whom I should contact regarding ARC toolchain? Thanks.

ruuddw, 2020-10-26:
That would be @abrodkin or @claziss.
Are there plans to use compiler managed thread local storage with Zephyr? That would require some more work than toolchain alone. Also, it could introduce gcc dependencies (not sure other toolchains like e.g. MetaWare supports it).

dcpleung, 2020-10-26:
TLS has been enabled on some architectures already @ https://github.com/zephyrproject-rtos/zephyr/pull/28982. Once toolchain supports generating TLS code, the arch code in Zephyr will also need to be updated to populate the thread pointer at context switch.

Not sure about MetaWare, but currently, TLS can be disabled via Kconfig. So we only need to enable TLS if there is toolchain support.

claziss, 2020-10-27:
ARC compiler supports TLS. When configuring the toolchain to use glibc or uclibc libraries (i.e., linux toolchain), the TP will automatically use r25 as the thread pointer. For bare metal, one needs to specify exactly which TP register (e.g. `-mtp-regno=r25`). However, this doesn't imply everything will work out of the box as TLS needs support in run-time libraries as well not only tools.

ruuddw, 2020-10-27:
Ah, I see work on TLS has been progressing and was merged few days ago - wasn't aware.
As  Claudiu explains, for the baremetal toolchain it is a matter of indicating what register to use as thread pointer. Guess r25 can be used, to be consistent with Linux  (or 26/27 as gp and fp aren't used, though it might be confusing).

evgeniy-paltsev, 2020-11-02:
@evgeniy-paltsev

dcpleung, 2020-11-09:
Use `-mtp-regno=r26` for TLS @ https://github.com/zephyrproject-rtos/zephyr/pull/29906
So this is no longer an issue.

closed 2020-11-09 by dcpleung