zephyrproject-rtos/sdk-ng #289: arm-cortex-m : DWARF error: could not find variable specification at offset xxx

giellamoswhard, 2020-11-06:
While building my project on arm-cortex-m withe 0.12-beta 1 I get something like 

/opt/toolchains/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 4398
/opt/toolchains/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 44ed
/opt/toolchains/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 451a
/opt/toolchains/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-objdump: DWARF error: could not find variable specification at offset 4544

Did some research got: 

https://sourceware.org/bugzilla/show_bug.cgi?id=26520

It mentions some patch

galak, 2020-11-06:
Do you have a test case / code you can share that shows the issue?

galak, 2020-11-08:
Should be fixed by https://github.com/zephyrproject-rtos/sdk-ng/pull/290

closed 2020-11-18 by galak