zephyrproject-rtos/sdk-ng #291: Cannot set up guest memory errors with ARC qemu targets

YuguoWH, 2020-11-09:
**Describe the bug**
When I follow the document to run qemu on ARC targets, an error pop up saying that it cannot set up guest memory 'arc.ram'. 
I ran across this error with both qemu_arc_em and qemu_arc_hs. 
I could run successfully with qemu_x86. 

**To Reproduce**
Steps to reproduce the behavior:
Follow the steps described in https://docs.zephyrproject.org/latest/boards/arc/qemu_arc/doc/index.html
1. west build -b qemu_arc_em samples/synchronization/
2. west build -t run

**Expected behavior**
samples/synchronization should be run on QEMU

**Logs and console output**
```
west build -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: arcem
qemu-system-arc: cannot set up guest memory 'arc.ram': Cannot allocate memory
FAILED: zephyr/CMakeFiles/run 
cd /home/yuguo/Documents/zephyr/build && /home/yuguo/zephyr-sdk-0.11.3/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-arc -cpu arcem -M simhs -m 8M -nographic -no-reboot -monitor none -global cpu.firq=false -global cpu.num-irqlevels=15 -global cpu.num-irq=25 -global cpu.ext-irq=20 -global cpu.freq_hz=1000000 -global cpu.timer0=true -global cpu.timer1=true -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -kernel /home/yuguo/Documents/zephyr/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/yuguo/.local/bin/cmake --build /home/yuguo/Documents/zephyr/build --target run
```

**Environment:**
 - OS: Ubuntu 18.04 LTS (64-bit), 2048MB RAM running on VM VirtualBox
 - Toolchain: Zephyr SDK 0.11.3
 - Zephyr master branch, Commit SHA 177ea931

**Addition:**
no matter how much RAM I changed with -m option, the error occurs. The issue is possibly caused due to limited RAM.

galak, 2020-11-09:
@abrodkin can you look into this

abrodkin, 2020-11-09:
That quite a funny problem happens because we [Ok, *me*, to be more precise] for the sake of simplicity decided to allocate all the usable memory for Linux on ARC use-case, i.e. 2 GiB in one go:
```c
#define SIMHS_RAM_BASE      0x80000000
#define SIMHS_RAM_SIZE      0x80000000
...
    /* Init system DDR */
    system_ram = g_new(MemoryRegion, 1);
    memory_region_init_ram(system_ram, NULL, "arc.ram", SIMHS_RAM_SIZE,
                           &error_fatal);
    memory_region_add_subregion(system_memory, SIMHS_RAM_BASE, system_ram);
```

That said all the hoops we do on QEMU instantiation with `-m 8M` make currently no sense. And on systems with limited amount of RAM that leads to inability to even start. And we never noticed that before because on modern systems we may easily allocate such a huge buffers, especially if they are not really used mostly, i.e. amount of real pages being used is just a fraction of what we assume to allocate.

Will come-up with an improvement soonish. @galak could you please assign this one to me?

galak, 2020-11-18:
@abrodkin any update on this?

abrodkin, 2020-11-23:
I tried a trivial fix I was thinking about:
```diff
diff --git a/hw/arc/sim-hs.c b/hw/arc/sim-hs.c
index 7dbf23bd03..61f50118c0 100644
--- a/hw/arc/sim-hs.c
+++ b/hw/arc/sim-hs.c
@@ -25,7 +25,6 @@
 #include "hw/sysbus.h"

 #define SIMHS_RAM_BASE      0x80000000
-#define SIMHS_RAM_SIZE      0x80000000
 #define SIMHS_IO_BASE       0xf0000000
 #define SIMHS_IO_SIZE       0x10000000
 #define SIMHS_UART0_OFFSET  0x0
@@ -49,7 +48,7 @@ static void simhs_init(MachineState *machine)
     int n;

     boot_info.ram_start = SIMHS_RAM_BASE;
-    boot_info.ram_size = SIMHS_RAM_SIZE;
+    boot_info.ram_size = machine->ram_size;
     boot_info.kernel_filename = machine->kernel_filename;
     boot_info.kernel_cmdline = machine->kernel_cmdline;

@@ -69,7 +68,7 @@ static void simhs_init(MachineState *machine)

     /* Init system DDR */
     system_ram = g_new(MemoryRegion, 1);
-    memory_region_init_ram(system_ram, NULL, "arc.ram", SIMHS_RAM_SIZE,
+    memory_region_init_ram(system_ram, NULL, "arc.ram", machine->ram_size,
                            &error_fatal);
     memory_region_add_subregion(system_memory, SIMHS_RAM_BASE, system_ram);
```
And even though result measured with `pmap PID | tail -n1` was sensibly smaller (1'538'712K vs 2'310'484K) I guess that doesn't bring a lot in a sense that for requested 8MiB of target RAM (with `-m 8M`) it is still way too much.

Now given we're in the middle of QEMU upstreaming (see https://lists.nongnu.org/archive/html/qemu-devel/2020-11/msg03101.html) I guess it makes not that much sense to spend any more time on this here, especially knowing that in many not that limited as in reported case environments it does work (even in Buildkite if I'm not mistaken).

@galak let me know what do you think of that, if you think we still should go after host memory footprint I may dig a bit deeper into it later.

FWIW if I build and run the same `../samples/hello_world/` example on `qemu_x86` I'm getting `pmap` reporting total of 263'000K. So there's definitely a room for improvement ;-)

evgeniy-paltsev, 2021-03-23:
@abrodkin should we move this issue to our ARC QEMU repo?

stephanosio, 2021-08-03:
> if you think we still should go after host memory footprint I may dig a bit deeper into it later.

When running sanitycheck, we can run multiple instances of qemu (up to max. core count), so 1.5GB per qemu inst can quickly add up (already 24GB for 16 cores). We really should look into reducing memory footprint.

abrodkin, 2021-08-03:
@cupertinomiranda do you know if there were any improvements related to QEMU for ARC run-time memory footprint?

abrodkin, 2021-08-05:
Thanks @cupertinomiranda for a productive discussion and 4-hand debugging session.

So the major contributor to that "insane" memory allocation is actually QEMU's code gen buffer of default size 1 GiB (see `DEFAULT_CODE_GEN_BUFFER_SIZE_1` in [accel/tcg/translate-all.c](https://git.qemu.org/?p=qemu.git;a=blob;f=accel/tcg/translate-all.c;h=b12d0898d0aa4ec5df3a6131116d6fee570d51c8;hb=609d7596524ab204ccd71ef42c9eee4c7c338ea4#l955)) statically allocated on start.

Note it's architecture agnostic, and so behavior is the same on all architectures & targets*.
And in fact if I try QEMU built from v6.0.0 then I see very similar results of `pmap` for ARC & ARM (32-bit).
Fortunately it's possible to tune that buffer size as easy as `-accel tcg,tb-size=128`.

And combined with proposed change (https://github.com/zephyrproject-rtos/sdk-ng/issues/291#issuecomment-732429132) for variable target memory allocation (setting it for example with `-m 8M`) we may really get down to 550 MiB for ARC. While with `virt` "board" for ARC as it is now we're still around 2.6 GiB.

Sadly enough in the worst-case (i.e. as it is now w/o changes) with QEMU 6.x we'd get up to 3.5 GiB per one instance:
 * 2 GiB allocated by ARC "virt" board for entire address space (from 0x8000_0000 up to the end of 32-bit addresses)
 * 1 GiB allocated by QEMU core for code gen buffer
 * 0.5 GiB allocated in small chunks by QEMU as well

It's unfortunate that SDK 0.13.0 was just released, but if it makes sense (and given all explanation above I do think it makes a lot of sense) I'll be willing to post a patch for QEMU in SDK-ng adding support of variable target memory alocation. @galak what do you think?

* And since I cannot recall which QEMU binary I used for x86 exercise (cited above with allocation of 263 MiB) I cannot figure out exactly why its memory allocation was that small.

galak, 2021-08-17:
Any update here?

abrodkin, 2021-08-23:
@galak is there a git tree with all the changes from https://github.com/zephyrproject-rtos/sdk-ng/tree/main/meta-zephyr-sdk/recipes-devtools/qemu/files?

https://github.com/zephyrproject-rtos/qemu seems to be quite outdated, @stephanosio's https://github.com/stephanosio/zephyr-qemu was also last updated a year ago.

stephanosio, 2021-08-23:
> @galak is there a git tree with all the changes from https://github.com/zephyrproject-rtos/sdk-ng/tree/main/meta-zephyr-sdk/recipes-devtools/qemu/files?

@abrodkin Not at the moment.

We will eventually create our own forks of all the components built by Yocto (including QEMU) because, as part of https://github.com/zephyrproject-rtos/zephyr/issues/37255, we need to be able to build these host tool components from multiple build environments outside of Yocto (e.g. deb packaging system, rpm packaging system, Homebrew bottling system, ...).

closed 2021-09-01 by galak