zephyrproject-rtos/sdk-ng #293: Coverage does not work with qemu targets on 0.12 pre-releases.

nashif, 2020-11-17:
with SDK 0.12, coverage does not work, i.e.

`sanitycheck --coverage -p qemu_x86 -T tests/kernel/semaphore/`

results in:
```
GCOV_COVERAGE_DUMP_STARTE: Page fault at address 0x4230323a (error code 0x0)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to read
E: PDPTE: not present
E: EAX: 0x4230322a, EBX: 0x00000000, ECX: 0x00000054, EDX: 0x00000000
E: ESI: 0x00000000, EDI: 0x0010b246, EBP: 0x001acf84, ESP: 0x001acf74
E: EFLAGS: 0x00000206 CS: 0x0008 CR3: 0x001bf000
E: call trace:
E: EIP: 0x0010ff8a
E:      0x00110520 (0x1afea0)
E:      0x00122474 (0x0)
E:      0x0010b279 (0x0)
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x001696e0 (main)
E: Halting system
```

this works fine with 0.11. Verified that this is a binary issue, not related to qemu.

nashif, 2020-11-17:
crashes only with userspace, with qemu_x86_nommu, no crash, but also no coverage... So it looks like coverage data is stored wrong!

andrewboie, 2020-11-17:
what is at `0x4230323a`

That looks like a bogus memory address, it's not even mapped in the page tables.

galak, 2020-11-18:
Looks like the issue is GCOV_COUNTERS differs on gcc-10.

Fix here:
https://github.com/zephyrproject-rtos/zephyr/pull/30122

closed 2020-11-19 by galak