zephyrproject-rtos/sdk-ng #296: qemu_cortex_m3 minimal example: fails with undefined reference to `_exit'

oberstet, 2020-11-21:
I am following the official Zephyr getting started and want to build and run the "minimal" example on qemu for x86-64 and aarch64.

It works for me on qemu x86-64, but fails with the following for Arm:

```
-- west build: making build dir /home/oberstet/scm/crossbario/st-device/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/oberstet/scm/crossbario/st-device/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/oberstet/scm/crossbario/st-device/zephyr/samples/basic/minimal
-- Zephyr version: 2.4.0 (/home/oberstet/scm/crossbario/st-device/zephyr)
-- Found Python3: /home/oberstet/cpy382_1/bin/python3 (found suitable exact version "3.8.2") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: qemu_cortex_m3
-- Cache files will be written to: /home/oberstet/.cache/zephyr
-- Found toolchain: zephyr (/home/oberstet/zephyr-sdk-12)
-- Found dtc: /home/oberstet/zephyr-sdk-12/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/oberstet/scm/crossbario/st-device/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts
-- Generated zephyr.dts: /home/oberstet/scm/crossbario/st-device/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/oberstet/scm/crossbario/st-device/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/oberstet/scm/crossbario/st-device/zephyr/Kconfig
Loaded configuration '/home/oberstet/scm/crossbario/st-device/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3_defconfig'
Configuration saved to '/home/oberstet/scm/crossbario/st-device/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/oberstet/scm/crossbario/st-device/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/oberstet/zephyr-sdk-12/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Error at ../../../cmake/extensions.cmake:1557 (message):
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  ../../../CMakeLists.txt:42 (assert)

-- Configuring incomplete, errors occurred!
See also "/home/oberstet/scm/crossbario/st-device/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "/home/oberstet/scm/crossbario/st-device/zephyr/build/CMakeFiles/CMakeError.log".
FATAL ERROR: command exited with status 1: /snap/bin/cmake -DWEST_PYTHON=/home/oberstet/cpy382_1/bin/python3 -B/home/oberstet/scm/crossbario/st-device/zephyr/build -S/home/oberstet/scm/crossbario/st-device/zephyr/samples/basic/minimal -GNinja -DBOARD=qemu_cortex_m3
Makefile:102: recipe for target 'build_minimal_arm' failed
make: *** [build_minimal_arm] Error 1
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ cat /home/oberstet/scm/crossbario/st-device/zephyr/build/CMakeFiles/CMakeError.log
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /home/oberstet/zephyr-sdk-12/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc 
Build flags: 
Id flags:  

The output was:
1
/home/oberstet/zephyr-sdk-12/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/oberstet/zephyr-sdk-12/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x60): undefined reference to `_exit'
collect2: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /home/oberstet/zephyr-sdk-12/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ 
Build flags: 
Id flags:  

The output was:
1
/home/oberstet/zephyr-sdk-12/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/oberstet/zephyr-sdk-12/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text.exit+0x60): undefined reference to `_exit'
collect2: error: ld returned 1 exit status
```

https://gist.github.com/oberstet/5cf96cde06d289b8d70ca3e59c7daca8

---

I have been trying different sdk version, however no luck with:

* sdk-0.12.0-beta-2
* sdk-0.11.4

I'm a bit lost, as the documented getting started doesn't work for me. anything I can try? any traps? thanks a lot!

galak, 2020-11-24:
Have you tried removing and creating the build dir?

galak, 2020-11-24:
@tejlmand @mbolivar-nordic does this look familiar?  (my other thought was something w/the zephyr cmake cache)

mbolivar-nordic, 2020-11-24:
I see 'undefined reference to _exit' in reports every once in a while. I don't know what's causing it these days, though.

In old mailing list threads about the same symptom it was usually due to an unset ZEPHYR_BASE, but that's not the problem in this report; the build output makes it clear that zephyr is found.

I'm curious what @tejlmand might know.

oberstet, 2020-11-24:
as suggested, I've been trying deleting the build and cache dirs

```
rm -rf ./zephyr/build
rm -rf /home/oberstet/.cache/zephyr
```

with no luck (https://gist.github.com/oberstet/2d6b0d85effdb0a301de9c9dbabd5907) :(

galak, 2020-11-24:
@oberstet hmm, wonder if you build by hand:

```
mkdir samples/basic/minimal/build
cd samples/basic/minimal/build
cmake -DBOARD=qemu_cortex_m3 ..
make
```

galak, 2020-11-24:
@oberstet what version of cmake are you using?

oberstet, 2020-11-24:
thanks for helping! I'm using cmake v3.19.0, tried what you suggested (manual build): same result https://gist.github.com/oberstet/0278214a8c8299e3b7bda44f4762bd8d

oberstet, 2020-11-24:
not sure if that is actually my problem, but following https://stackoverflow.com/a/43836115/884770, I've been trying to build with `--specs=nosys.specs` and with

```
cmake -DCMAKE_EXE_LINKER_FLAGS="--specs=nosys.specs" -DCMAKE_TRY_COMPILE_TARGET_TYPE="STATIC_LIBRARY" -DBOARD=qemu_cortex_m3 ..
```

same problem.

galak, 2020-11-24:
can you try updating to cmake 3.19.1 or downgrading?  Saw I comment that makes me think that could be the issue.

oberstet, 2020-11-24:
ok, I've built cmake from upstream latest master:

```
cmake version 3.19.20201124-g5b8b838
```

same problem: https://gist.github.com/oberstet/5f1ad1bcfef8b5a9e933a940e882ee27

oberstet, 2020-11-24:
So I'm kinda running out of options as the simplest example from the getting started doesn' work for me. Here is version summary of what I tried:

* GNU 10.2.0 (from zephyr-sdk-0.12.0-beta-2) (also tried GNU 9.2.0 from zephyr-sdk-0.11.4)
* Zephyr version: 2.4.0 
* cmake version 3.19.20201124-g5b8b838 (also tried 3.19.0)
* **samples/basic/minimal**

Does the Zephyr project have a Docker toolchain image that bundles everything into a known good? Is there anything else I can try?

galak, 2020-11-24:
What distro are you using?

Docker:
https://github.com/zephyrproject-rtos/docker-image

and the CI system uses:
https://github.com/zephyrproject-rtos/ci-dockerfiles

oberstet, 2020-11-24:
ok, I tried ^ :+1: 

I've built the toolchain image above (https://github.com/zephyrproject-rtos/docker-image with `tag==v0.13`).

using that, and mounting the exact Zephyr source tree (v2.4.0 release) from my host to within the toolchain container, I now can:

* run cmake successfully for both Arm and x86 (so Arm works within the toolchain docker image, but not on my host)
* run make successfully for x86 (but not Arm: "fatal error: core_cm3.h: No such file or directory")

https://gist.github.com/oberstet/c6ccf64a6ffb57ce26f3b10060245a8d
https://gist.github.com/oberstet/47ba8a53e00fec1584b3c2acad893a50

galak, 2020-11-24:
@oberstet for the non-docker version with the build by hand instructions.

Can you do:
`cmake -DBOARD=qemu_cortex_m3 --trace-expand ..`

galak, 2020-11-24:
> * run make successfully for x86 (but not Arm: "fatal error: core_cm3.h: No such file or directory")

you probably need to do a `west update` to get the modules pulled down.

oberstet, 2020-11-24:
```
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal$ rm -rf build/
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal$ mkdir build 
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal$ cd build/
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal/build$ cmake -DBOARD=qemu_cortex_m3 --trace-expand .. >> cmake.log 2>&1
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal/build$ 
```

[cmake.log](https://github.com/zephyrproject-rtos/sdk-ng/files/5592415/cmake.log)

oberstet, 2020-11-24:
> you probably need to do a west update to get the modules pulled down.

not sure .. I am confused:

```
oberstet@intel-nuci7:~/scm/crossbario/st-device$ docker run -ti -v ${PWD}/zephyr:/workdir zephyr_doc:v0.13
Openbox-Message: Unable to find a valid menu file "/var/lib/openbox/debian-menu.xml"
user@6ec6384deb8f:/workdir$ 
The VNC desktop is:      6ec6384deb8f:0
PORT=5900

user@6ec6384deb8f:/workdir$ cd samples/basic/minimal/
user@6ec6384deb8f:/workdir/samples/basic/minimal$ ls -la
total 56
drwxr-xr-x  4 user user 4096 Nov 24 20:14 .
drwxr-xr-x 10 user user 4096 Nov 21 13:26 ..
-rw-r--r--  1 user user   17 Nov 21 13:26 arm.conf
drwxr-xr-x  5 user user 4096 Nov 24 20:15 build
-rw-r--r--  1 user user  190 Nov 22 06:17 CMakeLists.txt
-rw-r--r--  1 user user  532 Nov 21 13:26 common.conf
-rw-r--r--  1 user user  143 Nov 21 13:26 mt.conf
-rw-r--r--  1 user user  100 Nov 21 13:26 no-mt.conf
-rw-r--r--  1 user user   32 Nov 21 13:26 no-preempt.conf
-rw-r--r--  1 user user   59 Nov 21 13:26 no-timers.conf
-rw-r--r--  1 user user 2356 Nov 21 13:26 README.rst
-rw-r--r--  1 user user 2031 Nov 21 13:26 sample.yaml
drwxr-xr-x  2 user user 4096 Nov 21 13:26 src
-rw-r--r--  1 user user   17 Nov 21 13:26 x86.conf
user@6ec6384deb8f:/workdir/samples/basic/minimal$ rm -rf build
user@6ec6384deb8f:/workdir/samples/basic/minimal$ mkdir build
user@6ec6384deb8f:/workdir/samples/basic/minimal$ west update
FATAL ERROR: no west workspace found from "/workdir/samples/basic/minimal"; "west update" requires one.
Things to try:
  - Change directory to somewhere inside a west workspace and retry.
  - Set ZEPHYR_BASE to a zephyr repository path in a west workspace.
  - Run "west init" to set up a workspace here.
  - Run "west init -h" for additional information.

user@6ec6384deb8f:/workdir/samples/basic/minimal$ cd ..
user@6ec6384deb8f:/workdir/samples/basic$ cd ..
user@6ec6384deb8f:/workdir/samples$ cd ..
user@6ec6384deb8f:/workdir$ cd ..
user@6ec6384deb8f:/$ cd ..
user@6ec6384deb8f:/$ cd workdir/
user@6ec6384deb8f:/workdir$ west update
FATAL ERROR: no west workspace found from "/workdir"; "west update" requires one.
Things to try:
  - Change directory to somewhere inside a west workspace and retry.
  - Set ZEPHYR_BASE to a zephyr repository path in a west workspace.
  - Run "west init" to set up a workspace here.
  - Run "west init -h" for additional information.

user@6ec6384deb8f:/workdir$ cd samples/
user@6ec6384deb8f:/workdir/samples$ west update
FATAL ERROR: no west workspace found from "/workdir/samples"; "west update" requires one.
Things to try:
  - Change directory to somewhere inside a west workspace and retry.
  - Set ZEPHYR_BASE to a zephyr repository path in a west workspace.
  - Run "west init" to set up a workspace here.
  - Run "west init -h" for additional information.

user@6ec6384deb8f:/workdir/samples$ 
```

galak, 2020-11-24:
You should have a `.west` dir in /

If not:
`cd /workdir`
`west init -l .`

than try:
`west update`

galak, 2020-11-24:
> `cmake -DBOARD=qemu_cortex_m3 --trace-expand ..`

Can you do:

```
rm -fr /home/oberstet/.cache/zephyr
cmake -DBOARD=qemu_cortex_m3 --trace-expand .. >> cmake.log 2>&1
```

oberstet, 2020-11-24:
thanks!

here:

rgd cmake trace:

```
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ make clean 
#-find . -type d -name build -exec rm -rf {} \;
rm -rf ./zephyr/build
rm -rf /home/oberstet/.cache/zephyr
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ cd zephyr/samples/basic/minimal/
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal$ rm -rf build/
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal$ mkdir build
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal$ cd build/
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal/build$ rm -fr /home/oberstet/.cache/zephyr
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal/build$ cmake -DBOARD=qemu_cortex_m3 --trace-expand .. >> cmake2.log 2>&1
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device/zephyr/samples/basic/minimal/build$ 
```

[cmake2.log](https://github.com/zephyrproject-rtos/sdk-ng/files/5592579/cmake2.log)

-----

rgd west:

```
oberstet@intel-nuci7:~/scm/crossbario/st-device$ docker run -ti -v ${PWD}/zephyr:/workdir zephyr_doc:v0.13
Openbox-Message: Unable to find a valid menu file "/var/lib/openbox/debian-menu.xml"
user@dd293499de1b:/workdir$ 
The VNC desktop is:      dd293499de1b:0
PORT=5900

user@dd293499de1b:/workdir$ echo "foo" > test.txt
user@dd293499de1b:/workdir$ cat test.txt 
foo
user@dd293499de1b:/workdir$ west init -l .
=== Initializing from existing manifest repository workdir
--- Creating /.west and local configuration file
FATAL ERROR: Cannot initialize in /.west: permission denied
user@dd293499de1b:/workdir$ 
```

galak, 2020-11-24:
On the docker side:

so I'd suggest the following (removal of the zephyr dir):
`docker run -ti -v ${PWD}:/workdir zephyr_doc:v0.13`

than:
```
cd /workdir/zephyr
west init -l .
west update
```

galak, 2020-11-24:
@oberstet I've been able to reproduce the non-docker case.

Seems cmake version related.  If I use:

```
[galak@spiff build]$ cmake --version
cmake version 3.18.4
```

things work.  Seems like anything 3.19 or newer fails.

oberstet, 2020-11-24:
yes, I can confirm: cmake version 3.18.5 indeed works (non-docker / on-host)!

thanks a lot for helping out! =)

https://gist.github.com/oberstet/ef829902d43a478cdb35027395e1ec6f

oberstet, 2020-11-24:
one more comment: users will have to compile cmake 3.18 from upstream on ubuntu bionic (which is what I used):

```
apt:
cmake/bionic-updates 3.10.2-1ubuntu2.18.04.1 amd64
  Plattformübergreifendes, quelloffenes Make-System

snap:
cmake                 3.19.0    crascit✓     classic   Cross-platform software build system generator
```

so apt is too old, and snap too fresh ...

pfalcon, 2020-11-24:
There were old good days when it was suggested to install cmake from https://pypi.org/project/cmake/ . And it still appears to have non-broken version...

oberstet, 2020-11-24:
^ fantastic! makes sense (and I wasn't aware .. thx!). love it. works:

```
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ make clean
#-find . -type d -name build -exec rm -rf {} \;
rm -rf ./zephyr/build
rm -rf /home/oberstet/.cache/zephyr
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ which cmake
/usr/local/bin/cmake
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ pip install cmake==3.18.4.post1
Collecting cmake==3.18.4.post1
  Downloading https://files.pythonhosted.org/packages/bb/c5/9a07d851bf8356dba2871f141f3e5e8d390995687be20316041a1e564d4c/cmake-3.18.4.post1-py3-none-manylinux1_x86_64.whl (17.7MB)
     |████████████████████████████████| 17.7MB 12.6MB/s 
Installing collected packages: cmake
Successfully installed cmake-3.18.4.post1
WARNING: You are using pip version 19.2.3, however version 20.2.4 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ which cmake
/home/oberstet/cpy382_1/bin/cmake
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ cmake --version
cmake version 3.18.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ make build_minimal_arm 
cd ./zephyr && \
	west build -p auto -b qemu_cortex_m3 samples/basic/minimal
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/oberstet/scm/crossbario/st-device/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/oberstet/scm/crossbario/st-device/zephyr/samples/basic/minimal
-- Zephyr version: 2.4.0 (/home/oberstet/scm/crossbario/st-device/zephyr)
-- Found Python3: /home/oberstet/cpy382_1/bin/python3 (found suitable exact version "3.8.2") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: qemu_cortex_m3
-- Cache files will be written to: /home/oberstet/.cache/zephyr
-- Found toolchain: zephyr (/home/oberstet/zephyr-sdk-12)
-- Found dtc: /home/oberstet/zephyr-sdk-12/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/oberstet/scm/crossbario/st-device/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3.dts
-- Generated zephyr.dts: /home/oberstet/scm/crossbario/st-device/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/oberstet/scm/crossbario/st-device/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/oberstet/scm/crossbario/st-device/zephyr/Kconfig
Loaded configuration '/home/oberstet/scm/crossbario/st-device/zephyr/boards/arm/qemu_cortex_m3/qemu_cortex_m3_defconfig'
Configuration saved to '/home/oberstet/scm/crossbario/st-device/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/oberstet/scm/crossbario/st-device/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/oberstet/zephyr-sdk-12/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/oberstet/scm/crossbario/st-device/zephyr/build
-- west build: building application
[1/116] Preparing syscall dependency handling

[111/116] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:        7852 B       256 KB      3.00%
            SRAM:        4016 B        64 KB      6.13%
        IDT_LIST:           8 B         2 KB      0.39%
[116/116] Linking C executable zephyr/zephyr.elf
(cpy382_1) oberstet@intel-nuci7:~/scm/crossbario/st-device$ 
```

pfalcon, 2020-11-24:
> (and I wasn't aware .. thx!)

In my mental list (and I'm long time with Zephyr), it should have been in the official getting started instructions. But before writing my previous comment, I checked the latest and a few earlier doc versions, and yep it's not there :-I. Well, I'd say most of Zephyr long-timers know about that "trick". Unless I really mix up things, I heard about it from @carlescufi, who might also comment whether that option was ever in the docs, but later removed, or my memory fails me, and what to do about it all now.

oberstet, 2020-11-24:
yes, that would have been useful for me;) as in, saved me a couple of hours (with the help of @galak that is). fwiw, I would actually suggest 2 improvements to the getting started:

1. "If it’s not version 3.13.1 or higher, follow these steps to add the Kitware third-party apt repository to get an updated version of cmake" => This is actually wrong: the version must be <3.19 as well. If that would have been there, it would have saved me as well
2. for Pythonistas, a `pip install cmake==3.18.4.post` is actually much better anyways .. better than apt or snap. and a venv is needed anyways ..

galak, 2020-11-24:
> 1. "If it’s not version 3.13.1 or higher, follow these steps to add the Kitware third-party apt repository to get an updated version of cmake" => This is actually wrong: the version must be <3.19 as well. If that would have been there, it would have saved me as well

Well the 3.19 release is pretty new (less than a week old) and I think we might actually be hitting a regression in CMake.  I've opened an issue to see what's going on.

galak, 2020-11-25:
CMake issue for 3.19 change to cmake_check_compiler_flag  - https://gitlab.kitware.com/cmake/cmake/-/issues/21497

closed 2020-11-25 by galak

galak, 2020-11-25:
Opened this issue to track 3.19 support

https://github.com/zephyrproject-rtos/zephyr/issues/30232