zephyrproject-rtos/sdk-ng #298: Newlib libc.a is compiled without optimization

martin-aberg, 2020-11-25:
The following applies to SDK 0.12.0-beta-2 and probably to the SDK 0.11 versions.

For `arm`, `sparc` and `riscv` it appears that:
- `libc.a` is compiled without optimization.
- `libc_nano.a` is compiled with optimization.

`libc.a` is also missing the `-g` option.

Only libraries for the architectures listed above have been investigated.

I would recommend that `libc.a` is compiled with optimization.

# Examples
Disassembly below is from the `zephyr-sdk-0.12.0-beta-2-x86_64-linux-setup.run` distribution.

## arm libc.a

```
Disassembly of section .text.unlink:

00000000 <unlink>:
   0:   e92d4800        push    {fp, lr}
   4:   e28db004        add     fp, sp, #4
   8:   e24dd008        sub     sp, sp, #8
   c:   e50b0008        str     r0, [fp, #-8]
  10:   e59f3020        ldr     r3, [pc, #32]   ; 38 <unlink+0x38>
  14:   e5933000        ldr     r3, [r3]
  18:   e51b1008        ldr     r1, [fp, #-8]
  1c:   e1a00003        mov     r0, r3
  20:   ebfffffe        bl      0 <_unlink_r>
  24:   e1a03000        mov     r3, r0
  28:   e1a00003        mov     r0, r3
  2c:   e24bd004        sub     sp, fp, #4
  30:   e8bd4800        pop     {fp, lr}
  34:   e12fff1e        bx      lr
  38:   00000000        .word   0x00000000
```

## arm libc_nano.a

```
Disassembly of section .text.unlink:

00000000 <unlink>:
   0:   e92d4010        push    {r4, lr}
   4:   e59f3010        ldr     r3, [pc, #16]   ; 1c <unlink+0x1c>
   8:   e1a01000        mov     r1, r0
   c:   e5930000        ldr     r0, [r3]
  10:   ebfffffe        bl      0 <_unlink_r>
  14:   e8bd4010        pop     {r4, lr}
  18:   e12fff1e        bx      lr
  1c:   00000000        .word   0x00000000
```

## sparc libc.a

```
Disassembly of section .text.unlink:

00000000 <unlink>:
   0:   9d e3 bf a0     save  %sp, -96, %sp
   4:   f0 27 a0 44     st  %i0, [ %fp + 0x44 ]
   8:   03 00 00 00     sethi  %hi(0), %g1
   c:   82 10 60 00     mov  %g1, %g1   ! 0 <unlink>
  10:   c2 00 40 00     ld  [ %g1 ], %g1
  14:   d2 07 a0 44     ld  [ %fp + 0x44 ], %o1
  18:   90 10 00 01     mov  %g1, %o0
  1c:   40 00 00 00     call  1c <unlink+0x1c>
  20:   01 00 00 00     nop
  24:   82 10 00 08     mov  %o0, %g1
  28:   b0 10 00 01     mov  %g1, %i0
  2c:   81 e8 00 00     restore
  30:   81 c3 e0 08     retl
  34:   01 00 00 00     nop
```

## sparc libc_nano.a

```
Disassembly of section .text.unlink:

00000000 <unlink>:
   0:   92 10 00 08     mov  %o0, %o1
   4:   03 00 00 00     sethi  %hi(0), %g1
   8:   d0 00 60 00     ld  [ %g1 ], %o0
   c:   82 13 c0 00     mov  %o7, %g1
  10:   40 00 00 00     call  10 <unlink+0x10>
  14:   9e 10 40 00     mov  %g1, %o7
```

## riscv libc.a

```
Disassembly of section .text.unlink:

00000000 <unlink>:
   0:   fe010113                addi    sp,sp,-32
   4:   00112e23                sw      ra,28(sp)
   8:   00812c23                sw      s0,24(sp)
   c:   02010413                addi    s0,sp,32
  10:   fea42623                sw      a0,-20(s0)
  14:   000007b7                lui     a5,0x0
  18:   0007a783                lw      a5,0(a5) # 0 <unlink>
  1c:   fec42583                lw      a1,-20(s0)
  20:   00078513                mv      a0,a5
  24:   00000097                auipc   ra,0x0
  28:   000080e7                jalr    ra # 24 <unlink+0x24>
  2c:   00050793                mv      a5,a0
  30:   00078513                mv      a0,a5
  34:   01c12083                lw      ra,28(sp)
  38:   01812403                lw      s0,24(sp)
  3c:   02010113                addi    sp,sp,32
  40:   00008067                ret
```

## riscv libc_nano.a

```
Disassembly of section .text.unlink:

00000000 <unlink>:
   0:   000007b7                lui     a5,0x0
   4:   00050593                mv      a1,a0
   8:   0007a503                lw      a0,0(a5) # 0 <unlink>

0000000c <.LVL1>:
   c:   00000317                auipc   t1,0x0
  10:   00030067                jr      t1 # c <.LVL1>
```

galak, 2020-11-25:
@martin-aberg a few comments:

* is nano libc build with `-g`?
* for the normal libc.a we made a choice to not optimize for size, with the assumption that its optimized for performance
* Do you now for libc.a if the build gets build with -O2 or -O3 or similar?

martin-aberg, 2020-11-25:
>     * is nano libc build with `-g`?
No.

I agree that it is a reasonable assumption that some optimization would be enabled by default by the newlib build scripts. Similar to how it is done with libstdc++.a for example. However, that is unfortunately not the case and for some reason Newlib/crosstools-ng defaults to _not_ give any `-O*` or `-g` option in the non-nano config.

Example from the build log file for the SDK 0.12-beta-2 riscv64 toolchain (https://buildkite.com/zephyr/sdk-ng/builds/205):

First for libc.a:
```
[ALL  ]    riscv64-zephyr-elf-cc -B/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/build/build-libc/riscv64-zephyr-elf/rv32i/ilp32/newlib/ -isystem /workdir/build/build_riscv64/.build/riscv64-zephyr-elf/build/build-libc/riscv64-zephyr-elf/rv32i/ilp32/newlib/targ-include -isystem /workdir/build/build_riscv64/.build/riscv64-zephyr-elf/src/newlib/newlib/libc/include -B/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/build/build-libc/riscv64-zephyr-elf/rv32i/ilp32/libgloss/riscv64 -L/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/build/build-libc/riscv64-zephyr-elf/rv32i/ilp32/libgloss/libnosys -L/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/src/newlib/libgloss/riscv64  -march=rv32i -mabi=ilp32 -DPACKAGE_NAME="newlib" -DPACKAGE_TARNAME="newlib" -DPACKAGE_VERSION="3.3.0" -DPACKAGE_STRING="newlib 3.3.0" -DPACKAGE_BUGREPORT="" -DPACKAGE_URL="" -I. -I/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/src/newlib/newlib/libc/reent -fno-builtin -DHAVE_NANOSLEEP -D__SINGLE_THREAD__ -D__NO_SYSCALLS__      -ffunction-sections -fdata-sections  -march=rv32i -mabi=ilp32 -c -o lib_a-unlinkr.o `test -f 'unlinkr.c' || echo '/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/src/newlib/newlib/libc/reent/'`unlinkr.c
```

and further down the corresponding compile line for libc_nano.a (same source file and multilib):

```
[ALL  ]    riscv64-zephyr-elf-cc -B/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/build/build-libc-nano/riscv64-zephyr-elf/rv32i/ilp32/newlib/ -isystem /workdir/build/build_riscv64/.build/riscv64-zephyr-elf/build/build-libc-nano/riscv64-zephyr-elf/rv32i/ilp32/newlib/targ-include -isystem /workdir/build/build_riscv64/.build/riscv64-zephyr-elf/src/newlib/newlib/libc/include -B/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/build/build-libc-nano/riscv64-zephyr-elf/rv32i/ilp32/libgloss/riscv64 -L/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/build/build-libc-nano/riscv64-zephyr-elf/rv32i/ilp32/libgloss/libnosys -L/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/src/newlib/libgloss/riscv64  -march=rv32i -mabi=ilp32 -DPACKAGE_NAME="newlib" -DPACKAGE_TARNAME="newlib" -DPACKAGE_VERSION="3.3.0" -DPACKAGE_STRING="newlib 3.3.0" -DPACKAGE_BUGREPORT="" -DPACKAGE_URL="" -I. -I/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/src/newlib/newlib/libc/reent -Os -fno-builtin -DHAVE_NANOSLEEP -D__SINGLE_THREAD__ -D__NO_SYSCALLS__      -ffunction-sections -fdata-sections -g -Os  -march=rv32i -mabi=ilp32 -c -o lib_a-unlinkr.o `test -f 'unlinkr.c' || echo '/workdir/build/build_riscv64/.build/riscv64-zephyr-elf/src/newlib/newlib/libc/reent/'`unlinkr.c
```

In summary:
- `libc_nano.a` is compiled with `-g -Os`.
- `libc.a` is compiled without  `-g` and without any `-O*` option.

galak, 2020-11-25:
@martin-aberg 

Going to try a local build with:

```
CT_LIBC_NEWLIB_TARGET_CFLAGS="-g -O2"
CT_LIBC_NANO_NEWLIB_TARGET_CFLAGS="-g"
```

for riscv64 and see how that works out.

martin-aberg, 2020-11-25:
Oops. I need to correct myself regarding nano and `-g`.

> * is nano libc build with `-g`?

Yes. it is.

martin-aberg, 2020-11-25:
Thank you @galak. I can confirm that adding

```
CT_LIBC_NEWLIB_TARGET_CFLAGS="-g -O2"
```
to `configs/sparc.config` solves the issue when building for SPARC (with `./go.sh sparc`).

galak, 2020-11-26:
So we purposely strip libs via `CT_STRIP_TARGET_TOOLCHAIN_LIBRARIES=y`

There was a tradeoff in size of the toolchain packages.

Added `-O2` for normal libc.a builds in #299

closed 2020-12-02 by galak