zephyrproject-rtos/sdk-ng #300: riscv: qemu needs SDK update to test memory protection

maksimmasalski, 2020-12-01:
**Describe the bug**
Most of the tests from memory protection directory can't be build for Qemu RISCV.

<b>Build failed on qemu_riscv32 and qemu_riscv64:</b>
`tests/kernel/mem_protect/futex/`
`tests/kernel/mem_protect/mem_map/`
`tests/kernel/mem_protect/mem_protect/`
`tests/kernel/mem_protect/obj_validation/`
`tests/kernel/mem_protect/syscalls/`
`tests/kernel/mem_protect/userspace/`

<b>Can build and execute on qemu_riscv32 and qemu_riscv64:</b>
`tests/kernel/mem_protect/protection/` -Test Fails
`tests/kernel/mem_protect/stack_random/` -Test Success
`tests/kernel/mem_protect/stackprot/` -Test Success
`tests/kernel/mem_protect/sys_sem/` -Test Success

**To Reproduce**
Steps to reproduce the behavior:
1. Build that tests one by one
west build -t run -p -b qemu_riscv64 tests/kernel/mem_protect/futex/
2. See error

**Expected behavior**
Tests can be build and executed successfully.

**Impact**
Can't test RISCV memory protection.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.4
 - Commit SHA or Version used cb1cf54406abafb4a781660161a93d5c363c7a1f

**Additional context**
Add any other context about the problem here.

galak, 2020-12-02:
Is this something more than what is already in SDK 0.12.0?

amergnat, 2020-12-03:
@galak I think it should work with SDK 0.12.0
@maksimmasalski It seems your PR haven't bug anymore. I supposed you find & fix the issue ?

maksimmasalski, 2020-12-04:
> @galak I think it should work with SDK 0.12.0
> @maksimmasalski It seems your PR haven't bug anymore. I supposed you find & fix the issue ?

That was not riscv related issue. Something in code base.

galak, 2021-01-20:
SDK 0.12.1 is now used by zephyr so closing this issue.

closed 2021-01-20 by galak