zephyrproject-rtos/sdk-ng #304: ZTEST with snprintf causes CPU fault on qemu_x86 (0.12.0-beta-3)

r2r0, 2020-12-03:
As in title.
0.12.0-beta-2 works well.
Zephyr commit HASH: c08ef381c81926483e1469df6972fe063ac25954
Below output log and sample ztest.

```
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..*** Booting Zephyr OS version 2.4.99  ***
Running test suite littlefs_tests
===================================================================
START - test_snprintf_bug
E: Floating point unit device not available
E: EAX: 0x00238e44, EBX: 0x0023f720, ECX: 0x0021e700, EDX: 0x0011adfe
E: ESI: 0x0023f720, EDI: 0x001093fd, EBP: 0x00238e20, ESP: 0x00238d28
E: EFLAGS: 0x00000246 CS: 0x0008 CR3: 0x00243000
E: call trace:
E: EIP: 0x001014f2
E:      0x001011f8 (0x23f720)
E:      0x00108c18 (0x238ec9)
E:      0x0010c941 (0x0)
E:      0x0010940c (0x23f2c0)
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x11d000 (unknown)
E: Halting system
```

```C
#include <ztest.h>

#define MAX_PATH_LEN   255

const char FS_MOUNTING_POINT[] = "/lfs";

static void test_snprintf_bug(void)
{
    char fname[MAX_PATH_LEN];
    //printk("before snprintf\r\n");
    snprintf(fname, sizeof(fname), "%s/test_file", FS_MOUNTING_POINT);
   // printk("after snprintf\r\n");

    zassert_true(1==1, "Test Ok");
}

void test_main(void)
{
    ztest_test_suite(littlefs_tests,
                     ztest_unit_test(test_snprintf_bug)
                     );

    ztest_run_test_suite(littlefs_tests);
}
```

```
CONFIG_ZTEST=y
CONFIG_ZTEST_STACKSIZE=32768

CONFIG_CONSOLE=y
CONFIG_SERIAL=y

CONFIG_NEWLIB_LIBC=y
```

galak, 2020-12-03:
Can you enable FLOATING POINT

galak, 2020-12-03:
The default x86 build for newlib ends up using floating point for things like initialization variables to zero.

@dcpleung we need to figure out how we want to than handle this:
* Introduce a non-fp multilib?
* Have FP enabled if newlib is enabled for x86
* other ideas

r2r0, 2020-12-03:
Adding CONFIG_FPU=y allows the test to pass.

galak, 2020-12-03:
@nashif "adding" you to see who else might want to chime in from an x86 point of view w/regards on how to handle floating point.

dcpleung, 2020-12-03:
That's weird. The changes between 0.12.0-beta-2 and 0.12.0-beta-3 should not affect how things are initialized. IIRC, x86 has multilib enabled with soft-fp. Was it disabled somehow?

pfalcon, 2020-12-03:
There were changes related to optimization flags for newlib, right? Could that cause more disturbance than expected?

galak, 2020-12-04:
> That's weird. The changes between 0.12.0-beta-2 and 0.12.0-beta-3 should not affect how things are initialized. IIRC, x86 has multilib enabled with soft-fp. Was it disabled somehow?

There is a soft-fp multilib, its just that we don't build with the soft-float flags by default so its not picked up.  The default x86 build is effectively hard-float right now.

dcpleung, 2020-12-05:
I wonder if the forced target cflags is overriding the cflags needed for correct x86 newlib build. Maybe do `CT_LIBC_NEWLIB_ENABLE_TARGET_OPTSPACE=n` in SDK instead? This is default to `y` in crosstool-ng.

galak, 2020-12-05:
> I wonder if the forced target cflags is overriding the cflags needed for correct x86 newlib build. Maybe do `CT_LIBC_NEWLIB_ENABLE_TARGET_OPTSPACE=n` in SDK instead? This is default to `y` in crosstool-ng.

The config change in the SDK is exposing the issue, but it has always existed.  The toolchain on x86 defaults to `-mhard-float` yet Zephyr's config does not match that assumption.  I'd argue that we should change the defaults on x86 to have `CONFIG_FPU` enabled by default.  At a minimum if `CONFIG_NEWLIB` is enabled since you aren't going to really know if various library code is using FP or not.

galak, 2020-12-09:
@dcpleung ping.

dcpleung, 2020-12-09:
> @dcpleung ping.

I am finishing up something else, so I should be able to look at this later today.

galak, 2020-12-09:
@dcpleung thanks.  Figuring out what we do here is I think the last item to resolve to make the 0.12.0 SDK release.

martin-aberg, 2020-12-10:
Here is a newlib patch which intends to avoid the floating point instruction in code paths where floating-point specifier is not used in the format string: https://github.com/martin-aberg/newlib-cygwin/commit/53ea91e9022bb129393721342aab3e55adba3c84 It has been verified with GCC 10.2.0 (SPARC) both by execution and inspection of disassembly. (Hit FPU load instruction for `snprintf("%s", s)` before patch applied but not after patch applied.)

I can not judge whether this approach is the right solution for Zephyr however.

galak, 2020-12-10:
@martin-aberg thanks for the patch.  However I feel like if we have a toolchain default to hard float than we should default the zephyr config to match that.  Since one never knows what library one may link and if its using hard float somewhere inside it or not.

dcpleung, 2020-12-10:
This https://github.com/zephyrproject-rtos/zephyr/pull/30620 should fix the issue without forcing `CONFIG_FPU=y`.

r2r0, 2020-12-17:
0.12.0-beta-4 produces the same error.

dcpleung, 2020-12-17:
> 0.12.0-beta-4 produces the same error.

Have you tried with https://github.com/zephyrproject-rtos/zephyr/pull/30620?

r2r0, 2020-12-17:
Yep - with [zephyrproject-rtos/zephyr#30620](https://github.com/zephyrproject-rtos/zephyr/pull/30620) my tests work again.

r2r0, 2020-12-21:
@galak 
Because [PR30620](https://github.com/zephyrproject-rtos/zephyr/pull/30620) is still not merged to Zephyr then official release for SDK 0.12.0 fails on our CI.

dcpleung, 2021-01-20:
https://github.com/zephyrproject-rtos/zephyr/pull/31230 is merged. Closing this one...

closed 2021-01-20 by dcpleung