zephyrproject-rtos/sdk-ng #306: Cannot link RISC-V 64 binaries with hardfloat

katsuster, 2020-12-05:
I'm using newest SDK (0.12.0-beta-3), it can build RISC-V 32 with hardfloat but cannot build RISC-V 64 with hardfloat.
It seems that this is not degrade of beta-3 because 0.12.0-beta-2 is also same.

### Environment

- Debian GNU/Linux testing (bullseye) on x86_64
- SDK 0.12.0-beta-3 (0.12.0-beta-2 is also same)

### How to reproduce

Add hardfloat support to qemu_riscv64 board:

```
--- a/boards/riscv/qemu_riscv64/Kconfig.board
+++ b/boards/riscv/qemu_riscv64/Kconfig.board
@@ -6,3 +6,4 @@ config BOARD_QEMU_RISCV64
        depends on SOC_RISCV_SIFIVE_FREEDOM
        select QEMU_TARGET
        select 64BIT
+       select CPU_HAS_FPU
```

And simply build it:

```
cmake -G Ninja -DBOARD=qemu_riscv64 ../tests/kernel/fpu_sharing/float_disable/
ninja
```

### Result

```
[112/117] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf
: && ccache /home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin
/riscv64-zephyr-elf-gcc   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.
c.obj -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=/home/katsuhiro/share/projects/oss/zephyr/zephyr/build32/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/riscv/core/libarch__riscv__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj  -L"/home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0"  -L/home/katsuhiro/share/projects/oss/zephyr/zephyr/build32/zephyr  -lgcc  -Wl,--print-memory-usage  zephyr/arch/common/libisr_tables.a  -mabi=lp64f  -march=rv64imafc  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && :
/home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: /home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/libgcc.a(_clzdi2.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf32-littleriscv' does not match `elf64-littleriscv'
/home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: failed to merge target specific data of file /home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/libgcc.a(_clzdi2.o)
/home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: /home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/libgcc.a(_ctzdi2.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf32-littleriscv' does not match `elf64-littleriscv'
/home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: failed to merge target specific data of file /home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/libgcc.a(_ctzdi2.o)
/home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: /home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/libgcc.a(_clz.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf32-littleriscv' does not match `elf64-littleriscv'
/home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/../../../../riscv64-zephyr-elf/bin/ld: failed to merge target specific data of file /home/katsuhiro/share/projects/oss/zephyr-sdk/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/libgcc.a(_clz.o)
```

galak, 2020-12-05:
So the issue here is that we don't support a 64-bit single precision only multilib.  The configuration is asking for `rv64imafc`.  We support `rv64imafdc` from the toolchain.  (assumption is that most 64-bit cores that support floating point will have both `f` and `d`.

Additionally the qemu model from what I can tell as configured doesn't support FP at all, since we are configure qemu to a SiFive E51 core:

https://github.com/qemu/qemu/blob/v5.1.0/target/riscv/cpu.c#L153

Which would be `rv64imacu`.

Are you really wanting support for single-precision only with 64-bit?

katsuster, 2020-12-05:
Ah sorry, I didn't check rv64imafdc. I understand the situation of RV64 toolchain. Thanks.
Answer is currently, No I do not want. There is no rv64imafc real/virtual processor in the market.

Yes, I know E31 and E51 cores do not support hardfloat. But now I've sent PR about RV32/RV64 virt boards.
If these PR are merged, we will get environments which support RV32/RV64 hardfloat.

At last, this is just interest. If such processor will appear, can we support rv64imacf in the future?

galak, 2020-12-08:
> At last, this is just interest. If such processor will appear, can we support rv64imacf in the future?

if whomever creates such a thing is willing to look at adding the changes to the toolchain to build that variant of the mutltilib.

closed 2020-12-08 by galak