zephyrproject-rtos/sdk-ng #311: aarch64-zephyr-elf-gdb: no TUI supported

carlocaione, 2020-12-15:
The AArch64 GDB doesn't support TUI mode:
```
# aarch64-zephyr-elf-gdb -tui
Cannot enable the TUI: error opening terminal [TERM=tmux-256color]
```

galak, 2020-12-15:
GDB TUI support isn't supported on any GDB variant in the SDK.  This is a limitation of crosstool-ng.  Feel free to submit a patch to upstream crosstool-ng to add support.

carlocaione, 2020-12-15:
This looks something it could help https://github.com/crosstool-ng/crosstool-ng/pull/575

galak, 2020-12-15:
> This looks something it could help [crosstool-ng/crosstool-ng#575](https://github.com/crosstool-ng/crosstool-ng/pull/575)

That should in theory be incorporated in our crosstool-ng version already.

stephanosio, 2020-12-15:
TUI should be working (minus source highlight).

https://github.com/zephyrproject-rtos/crosstool-ng-old/issues/37

Does it fail for the AArch64 GDB only? or the rest as well?

carlocaione, 2020-12-15:
Uhm, for me it's failing also for `arm-zephyr-eabi-`

galak, 2020-12-15:
> Uhm, for me it's failing also for `arm-zephyr-eabi-`

its not supported on any toolchain variant at this point.

stephanosio, 2020-12-15:
Try running it outside tmux. Also try `export TERM=xterm` or some variation of that.

carlocaione, 2020-12-15:
Yes, using `TERM=xterm-256color` it works. Shamefully closing this. Thanks!

closed 2020-12-15 by carlocaione