zephyrproject-rtos/sdk-ng #314: Add mips-zephyr-elf target support

frantony, 2020-12-22:
Please, add mips-zephyr-elf target support, in preparation for MIPS port, see https://github.com/zephyrproject-rtos/zephyr/pull/30405 for details.

galak, 2020-12-22:
Addressed by #315

galak, 2020-12-22:
@frantony can you try this build and see if the mips toolchain works for you:

builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/315/zephyr-sdk-0.12.0-pr-315-x86_64-linux-setup.run

frantony, 2020-12-23:
@galak 

> can you try this build and see if the mips toolchain works for you:
>
> http://builds.zephyrproject.org/zephyrproject-rtos/sdk-ng/315/zephyr-sdk-0.12.0-pr-315-x86_64-linux-setup.run

zephyr-sdk-0.12.0-pr-315 mostly works fine, thanks!

Please add this to the https://github.com/zephyrproject-rtos/sdk-ng/pull/315
```
--- a/cmake/zephyr/target.cmake
+++ b/cmake/zephyr/target.cmake
@@ -7,7 +7,7 @@ else()
 endif()
 set(CROSS_COMPILE_TARGET_nios2     nios2-zephyr-elf)
 set(CROSS_COMPILE_TARGET_riscv   riscv64-zephyr-elf)
-set(CROSS_COMPILE_TARGET_mips     mipsel-zephyr-elf)
+set(CROSS_COMPILE_TARGET_mips       mips-zephyr-elf)
 set(CROSS_COMPILE_TARGET_xtensa   xtensa-zephyr-elf)
 set(CROSS_COMPILE_TARGET_arc         arc-zephyr-elf)
 set(CROSS_COMPILE_TARGET_x86      x86_64-zephyr-elf)
```

Comparing with gccmipsmti toolchain several tests (5%) have build issues but I can fix it later.
At the moment I have this:

```
INFO    - 255 of 348 test configurations passed (73.28%), 93 failed, 981 skipped with 0 warnings in 501.11 seconds
INFO    - In total 2476 test cases were executed, 11316 skipped on 2 out of total 319 platforms (0.63%)
INFO    - 197 test configurations executed on platforms, 151 test configurations were only built.
```

So I'm ready to drop gccmipsmti toolchain in favour of zephyr toolchain.

galak, 2021-01-20:
Released SDK 0.12.1 with MIPS support

closed 2021-01-20 by galak