zephyrproject-rtos/sdk-ng #337: RISCV64: undefined reference to 'ffs' when calling __builtin_ffs()

dcpleung, 2021-04-24:
On SDK 0.12.4, when using `find_lsb_set()` (thus calling `__builtin_ffs()`), GCC complains about undefined reference to 'ffs'. This happens only on RISCV64, and does not affect RISCV32.

galak, 2021-05-05:
I think this is expected behavior.  Not 100% sure why.

See https://sourceware.org/pipermail/newlib/2017/014955.html

dcpleung, 2021-05-05:
x86-64 and AArch64 are working fine without this kind of workaround. It seems to be an issue on RISCV64 only.

galak, 2021-05-05:
> x86-64 and AArch64 are working fine without this kind of workaround. It seems to be an issue on RISCV64 only.

Yeah, asking the riscv compiler guys about this.

galak, 2021-06-02:
In discussion with risc-v compiler guys the determined this is expected behavior with gcc.

closed 2021-06-02 by galak