zephyrproject-rtos/sdk-ng #339: esp32 toolchain is missing some components

AWhiteKnight, 2021-05-02:
following the Getting Started on Ubuntu 21.01,installing the toolchain 0.12.4 with
`./zephyr-sdk-0.12.4-x86_64-linux-setup.run -- -d ~/zephyr-sdk-0.12.4
`
shows this output:
`...
 [*] Installing x86_64 tools... 
 [*] Installing xtensa_sample_controller tools... 
 [*] Installing xtensa_intel_apl_adsp tools... 
 [*] Installing xtensa_intel_s1000 tools... 
 [*] Installing xtensa_intel_bdw_adsp tools... 
 [*] Installing xtensa_intel_byt_adsp tools... 
 [*] Installing xtensa_nxp_imx_adsp tools... 
 [*] Installing xtensa_nxp_imx8m_adsp tools... 
 [*] Installing CMake files... 
 [*] Installing additional host tools... 
Success installing SDK.
SDK is ready to be used.
`
but when compiling some of the samples (basic/blinky, hello_world, esp32/wifi_station) I always get 
`...
 CMake Error at /home/stephan/zephyrproject/zephyr/cmake/compiler/gcc/target.cmake:10 (message):
  C compiler
  /home/stephan/zephyr-sdk-0.12.4/xtensa//xtensa-zephyr-elf/bin/xtensa-zephyr-elf-gcc
  not found - Please check your toolchain installation
...
`
The directory <sdk_path>/xtensa/xtensa-zephyr-elf does not exist, As it is not mentioned in the installation output, I think it is missing in the installation script (maybe together with some other espressif tools?).

galak, 2021-05-03:
The Zephyr xtensa toolchain does NOT support esp32.  You can see this here:

https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/xtensa/esp32/esp32.yaml

closed 2021-05-03 by galak

AWhiteKnight, 2021-05-03:
Why then does the command
`west build -p auto -b esp32 samples/basic/blinky`
try to use this toolchain?

galak, 2021-05-04:
This is because west / the zephyr build system doesn't validate the toolchain you attempt to use.