zephyrproject-rtos/sdk-ng #345: Consider disabling transactional memory support for libgcc

stephanosio, 2021-06-05:
The experimental [transactional memory support in the libgcc runtime](https://gcc.gnu.org/wiki/TransactionalMemory) is currently enabled by the Zephyr SDK (this feature is enabled by default, unless `USE_TM_CLONE_REGISTRY` is explicitly set to `0`).

Since this feature is currently not and will likely be never used by the Zephyr RTOS and its applications which target the embedded platforms with limited resources, consider disabling this feature.

This feature can be disabled by specifying `--disable-tm-clone-registry` (GCC 10+ only).

Note that GNU Arm Embedded also explicitly disables this feature. From build script:
```bash
$SRCDIR/$GCC/configure --target=$TARGET \
    --prefix=$INSTALLDIR_NATIVE \
    --libexecdir=$INSTALLDIR_NATIVE/lib \
    --infodir=$INSTALLDIR_NATIVE_DOC/info \
    --mandir=$INSTALLDIR_NATIVE_DOC/man \
    --htmldir=$INSTALLDIR_NATIVE_DOC/html \
    --pdfdir=$INSTALLDIR_NATIVE_DOC/pdf \
    --enable-languages=c,c++ \
    --enable-plugins \
    --disable-decimal-float \
    --disable-libffi \
    --disable-libgomp \
    --disable-libmudflap \
    --disable-libquadmath \
    --disable-libssp \
    --disable-libstdcxx-pch \
    --disable-nls \
    --disable-shared \
    --disable-threads \
    --disable-tls \
    --with-gnu-as \
    --with-gnu-ld \
    --with-newlib \
    --with-headers=yes \
    --with-python-dir=share/gcc-arm-none-eabi \
    --with-sysroot=$INSTALLDIR_NATIVE/arm-none-eabi \
    $GCC_CONFIG_OPTS                                \
    "${GCC_CONFIG_OPTS_LCPP}"                              \
    "--with-pkgversion=$PKGVERSION" \
    ${MULTILIB_LIST}

# Passing USE_TM_CLONE_REGISTRY=0 via INHIBIT_LIBC_CFLAGS to disable
# transactional memory related code in crtbegin.o.
# This is a workaround. Better approach is have a t-* to set this flag via
# CRTSTUFF_T_CFLAGS
make -j$JOBS CXXFLAGS="$BUILD_OPTIONS" \
        INHIBIT_LIBC_CFLAGS="-DUSE_TM_CLONE_REGISTRY=0"
```

**TODO:**
1. Add a crosstool-ng config for `--disable-tm-clone-registry`
2. Set the above config in the sdk-ng config files.

stephanosio, 2021-06-05:
cc @galak @tejlmand

galak, 2021-06-05:
@stephanosio seems reasonable, want to send a patch to upstream crosstool-ng for this.

stephanosio, 2021-06-05:
> want to send a patch to upstream crosstool-ng for this

yep, I will do that after I finish some local testing.

PR #347 (2021-06-16): crosstool-ng: Update to pull in gcc TM clone registry config
Update build script to pull in the crosstool-ng commit that supports
the new `CT_CC_GCC_TM_CLONE_REGISTRY` config option.

This option is disabled by default and will therefore add
`--disable-tm-clone-registry` during gcc build to disable the unused
transactional memory clone registry in libgcc.

For more details, refer to the issue #345.

Fixes #345.
closed 2021-06-16 by galak